Ultimaker/Cura #1471: Custom Machine problem (2.4.0)

MorganTHind, 2017-02-23:
Hi 
There is a few bug with the Custom machine settings. 
I configure a custom machine with my dimension but when I switch to ultimaker to my other machine, the build plate become a lot reduce ( 100x100x100 and not 265x230x160).
When I create a material with my custom machine, if i closed my software and open it a few moment later, my material dissapears for my machine BUT I can found them with my ultimaker.
Have a nice day.

Ghostkeeper, 2017-02-24:
I can't reproduce your first problem. These were my steps:

* Add an Ultimaker 3 printer.
* Add a Custom FDM Printer.
* In the dialogue that appears for Custom FDM Printer, set the width to 333mm.
* Switch to Ultimaker 3.
* Switch back to Custom FDM Printer.

The printer now has a width of 333mm. Seems fine.

The other issue (materials disappearing) is a duplicate of #1104, #1404, #1410 and #1452.

MorganTHind, 2017-02-28:
Sorry for it. 
I tried to make it again but i can't reproduce my problem. 
Thanks for the material disappearing solution.

closed 2017-07-27 by nallath