Ultimaker/Cura #1475: version 2.4 print area vs bed size is wrong

xreymason, 2017-02-24:
I've created a custom printer using the ultimate Original Dual as a starting point and the print bed is the right size but the area I can print in is smaller than it should be. How should I fix this or is this something Ultimaker will need to fix in their software? When I used this same setup in 2.3.1 it worked without this issue.
[image]

nallath, 2017-02-27:
Did you change the offset of the nozzles? the area on the left is caused by this.

xreymason, 2017-02-27:
You're right, I did. But from what I can tell on my printer it should be able to print X>=27mm and X<=285mm. Though from the looks of this it says more like X>=40or50mm and X<=285. Either way good point, I had the nozzle offset on the older version too and it didn't do this, that's why I was confused.

Thank you,
Xreymason

closed 2017-02-27 by xreymason

fieldOfView, 2017-02-27:
Protip: every square in the buildplate checkerboard is 10mm. So it looks like 47mm