Ultimaker/Cura #1476: "New Project" missing from "File" dropdown

tzminversion, 2017-02-25:
I think these commands dont need much explaination as they are pretty standard in any tool.

- New Project
- Save Project As...
- Close Project
- Reload Project from Disk

Otherwise I have to close the application to start a new project right now.

fieldOfView, 2017-02-26:
What would New Project do apart from clearing the buildplate, which you can already do with Edit -> Clear Build Plate?

What would be the difference between New Project and Close Project?

tzminversion, 2017-02-26:
I see what you mean. However, you have to know that first right? I didn't. I don't find it very intuitive.

1. I would expect New Project to start a fresh project with all settings set to their defaults.

2. Clearing the build plate would not reset to default settings.

3. Save Project currently should be called "Save Project As..." as it always opens the file dialog and it even says "Save to file" in the title of the dialog window.

4. Save Project should in theory save the currently open project.

5. Indeed, close project would only make sense if you had a welcome screen with recently opened projects available.

Ghostkeeper, 2017-03-01:
I've made an issue in our internal tracker to have this looked at by a UX designer. But don't mind me, please keep up the good discussion :)

I agree that "New Project" would make the workflow with projects more intuitive. It would then also need to ask if you'd like to save the old project. To make it even better, display the current project name somewhere, for example in the title bar of the window.

Ghostkeeper, 2017-03-15:
This has been added for the 2.6 release. Some pieces of it have already been implemented.

nallath, 2017-03-15:
This is another reason why we should have a better way to share nightly builds.

closed 2017-07-27 by nallath