Ultimaker/Cura #1478: Don't restart Cura when USB is connected!!

foreachthing, 2017-02-28:
- Updated Firmware on UM2
- Started Job from SD
- Closed Cura 2.4
- ... Job's almost done! Just one more hour.
- started Cura 2.4 (left USB cable attached) ..... ***BAM*** Printer stopped!
Thank you, Cura 2.4!

fieldOfView, 2017-02-28:
Sorry about your print!

Better rule of thumb: don't keep the usb cable connected when you don't use it.

foreachthing, 2017-03-01:
Yeah, me too! Just thought this should be documented somewhere or possibly caught in Cura, when the printer is printing, don't try to connect.

nallath, 2017-03-01:
We don't know that the printer is printing, untill we connect. It's actually a bit of a weird arduino issue that trying to connect to it causes a reboot.

fieldOfView, 2017-03-01:
So the only reasonable course of action is not to connect on startup, but only when a connection is required:
* when starting a print and/or switching to the print monitor
* when uploading firmware

From a UX perspective, it could be interesting to have an option to have Cura warn you before it tries to open the USB connection.

fieldOfView, 2017-03-01:
Also see this discussion/proposal:
https://github.com/Ultimaker/Cura/issues/1049

foreachthing, 2017-03-02:
I like that one: "without thinking"! :-)
Well, **now** I know!

ianpaschal, 2018-02-21:
Since this issue is almost a year old and hasn't had any activity in 11 months, I'm going to go ahead and close it. See fieldOfView's discussion which is open.

Ghostkeeper, 2018-03-13:
This is still an issue with the most recent version. If you start Cura with the USB cable connected to your printer, it ruins your print because Cura causes the printer to restart.

LipuFei, 2018-05-25:
@ChrisTerBeke That sounds like a nice feature, but a lot more work than expected... I will reopen this and make a new JIRA ticket.

fieldOfView, 2018-05-25:
@LipuFei, see #1049 for a proposal of the UI. I once got half implementing it in #1688 but other things (and the release schedule) got in the way. I could pick it up again around the end of june.

closed 2018-11-26 by Appesteijn

goscicki, 2019-05-16:
PLEASE make Cura stop connecting on startup. It still does this in 4.0. Why can't it just connect you click print? I keep opening a new window to tweak what I'm going to print next and lose hours of work/materials!

MollyUrs, 2020-01-07:
I can't believe this is still an issue, it also causes issues with loads of programs as it hogs all of the COM ports so I can't use my Arduino while printing :/