Ultimaker/Cura #1480: Octoprint plugin broken?

StefanKi, 2017-02-28:
Usually I saved my sliced files directly to Octoprint over the octoprint plugin.
Since some time the option is gone.
I also can't configure the Octoprint on my Custom FDM Printer preference screen.

I am using ppa:thopiekar/cura-master.
Am I doing something wrong? Has the configuration been moved?

fieldOfView, 2017-02-28:
I must admit I don't know the first thing about how ppas work. Did you update Cura to version 2.4? Then it is important that the OctoPrint plugin is also updated to the 2.4 version

StefanKi, 2017-02-28:
Thank you!
You brought me on the right way to solve the problem.
When updating from 2.3.99 to 2.4.99 there went something wrong i think.

The plugins were renamed!

In version 2.3.99 the plugins were called "cura-extra-plugin-..."
In version 2.4.99 the plugins are called "cura-plugin-..."

So of course the "cura-extra-plugin-octoprint" was held at version 2.3.99 and not updated to "cura-plugin-octoprint" for version 2.4.99.

I remove every packages named "cura-extra-plugin-..." and installed the corresponding versions of "cura-plugin-..." and now everything is working fine again.

closed 2017-02-28 by StefanKi

fieldOfView, 2017-02-28:
Perhaps @thopiekar can chime in with the reason for this change.

thopiekar, 2017-02-28:
Yes, the reason is quite simple. At the beginning I decided to call all extra plugins `cura-extra-plugin`, like when you would talk about it `application-attribute-noun`, but after a time, eg. the UM3 networking plugin, was part of Cura itself. Thus following my convention the name of the package also changed.
Therefore I decided to get a rid of the "-extra" in the package names to avoid any future conflicts and let the meta package cura-plugins-extra select all extra (incl. 3rd party) plugins.

I hope there won't be any conflicts in the future anymore :roll_eyes: