Ultimaker/Cura #1482: CURA 2.4 Prime tower not correct when machine_center_is_zero

dagnall53, 2017-03-02:
Reported before, but issues when machine centre is zero have  not been fully corrected in 2.4 .

When machine is set for machine_center_is_zero = true, the Prime tower "shadows" in the bottom left quadrant, but tries to print in the top right.  Causes all sorts of issues with the checks for disallowed regions and print bed settings!! 

Easy to test. ... Just change an example  Json to set machine centre is zero = true.. I used "punchtec_connect_xl.def.json", but I expect any dual extruder JSON would do. 

Not sure if the similarly wrongly interpreted "disallowed areas" issue that also came up with machine centre is zero has been fixed, but might be related..?

Thanks
Dagnall

fieldOfView, 2017-03-02:
Aww, I thought I had fixed all those issues. The problem remains that I don't actually have a delta, so I don't notice these things in daily life.

dagnall53, 2017-03-02:
Huge thanks for looking into this. The print tower issue is certainly not fixed.. But I have not actually checked the disallowed areas.. I will do that tomorrow.. Hopefully you had fixed that bit : -) 
I had assumed that since the one issue was not fixed, the related one was not either..

karabas, 2017-03-09:
fieldOfView, you don't need a delta for testing. Any 2extruder printer setting will show  the bug as I posted a pic https://cloud.githubusercontent.com/assets/67457/21853613/e9290c16-d828-11e6-911c-e2c0ce234377.png

fieldOfView, 2017-03-09:
@karabas: What I meant was, I don't have a delta so I don't actually use this functionality.

I contribute to Cura because I use the software myself. If I run into issues, I try to fix them. I don't use this functionality, so I don't run into issues using it. I'm not on a payroll.

I contributed the circular buildplate code, because many people were asking for it. But because I don't use it myself, I have to rely on others to test it.

karabas, 2017-03-09:
Accepted

dagnall53, 2017-03-10:
@fieldOfView  Many thanks for the clarification .. I was very pleased to see the Elliptical build plate, as it hugely helps when using a delta. In fact it means that most of the need for weird disallowed areas has probably gone away.  
However it does raise an interesting issue. The  "center= zero" issues are not unique to the elliptical build plate, they occur with rectangular plates just the same, so do we need someone else to  take up the challenge?

fieldOfView, 2017-03-10:
If someone else wants to pick up the issue, that would be great. As I said, I am willing to take a look but I don't have time the coming weeks so I will miss the window for the Cura 2.5 release.

> The "center= zero" issues are not unique to the elliptical build plate, they occur with rectangular plates just the same

I know. But the issue still remains that my printers don't need "center = zero", so I do not notice this defect. Also the issue is currently limited to center=zero in combination with multi-extrusion, which is an even smaller "market".

dagnall53, 2017-03-10:
Thanks. I appreciate testing these code issues is certainly frustrating.. so finding the "smaller market" issues will take time. 
I am persevering with Cura 2.4 because I think the interface has benefits, and certainly for multi extrusion, the new interface is nicer. But the way that one needs to reload the program to update G code in the jsons is extremely frustrating...
And just FYI.. I just removed the "elliptic" definition from my machine to test if the center =zero issue was unique to elliptic or was also there for rectangular, (it is) and even after replacing the code machine shape = elliptic in my json, Cura is now showing me a rectangular build plate....grr  
I guess its storing/hiding stuff in appdata as well... VERY frustrating.  I will delete the appdata directory and start again....

dagnall53, 2017-03-10:
FYI deleting the appdata local cura directory brought back the circular build plate when I reinstalled the printer..

nallath, 2017-03-10:
We're making the Cura 2.5 build beta as we speak...

dagnall53, 2017-03-10:
@nallath 
Has the centre =zero been fixed in 2.5? If so, I will download 2.5 when it appears and check it out..!.. 
Dagnall

PS, I have put some stuff about the dual extruder on thingyverse: http://www.thingiverse.com/thing:2166872

Ghostkeeper, 2017-03-10:
It'll take some time before the 2.5 beta is released to the public though because the System Testing team has to test its integration with other Ultimaker products first.

Ghostkeeper, 2017-03-10:
I just tested this issue on 2.5 using Punchtec XL by modifying `machine_center_is_zero` to `true` in its definition. The default position for the prime tower is marked red then, but that is to be expected. When I change the position of the prime tower to 100,100 it slices and puts the prime tower exactly on top of the shadow.

So it seems to work in the 2.5 beta.

fieldOfView, 2017-03-10:
Ah, I see now. I made the fix "so long ago" (in januari) I did not realize/remember it was not fixed in time for 2.4. The fix was merged into the master branch after the 2.4 branch was created:
https://github.com/Ultimaker/Cura/pull/1336

closed 2017-07-23 by BagelOrb