Ultimaker/Cura #1483: memory usage on macOS 10.12

waspinator, 2017-03-03:
After running for a few hours, Activity Monitor reports Cura 2.4 using more than 60GB of memory.

Ghostkeeper, 2017-03-06:
It typically stays around 2GB here on Ubuntu. Seems this is a memory leak specifically in MacOS then.

Ghostkeeper, 2017-06-05:
Could it be happening only after connecting to a UM3 printer and then seeing the camera image in the print monitor?

Ghostkeeper, 2017-06-05:
Our product manager just closed this for being inactive for 3 months.

waspinator, 2017-06-15:
http://i.imgur.com/4AFkCBzl.jpg

ianpaschal, 2018-02-27:
I'm not experiencing this (I typically have at least 2 instances of Cura open all day), and there's no more recent reports of it, so I'm closing this. It was likely fixed in later versions.

closed 2018-02-27 by ianpaschal

Ghostkeeper, 2018-03-19:
We've whack-a-moled a bug that was causing this during versions 2.7 and 3.0 if you've accessed the print monitor. The camera image wasn't properly closing when the tab was closed, and a bug in Qt causes the camera data to accumulate for frames that it misses due to a glitchy network connection.