Ultimaker/Cura #1485: Cura 2.4: Project's file extension

foreachthing, 2017-03-03:
The extension `.curaproject.3mf` is *always* added to any project saved. If I save a project, click on an existing project (i.e. TestProject) it is saved as `TestProject.curaproject.3mf.curaproject.3mf` instead if overwriting the `TestProject.curaproject.3mf`.

nallath, 2017-03-03:
I've made a number of changes to this in 2.5, so it should be fixed

Ghostkeeper, 2017-03-06:
Using OSX, are you?

foreachthing, 2017-03-06:
@Ghostkeeper : no, Windows 7, 8 and 8.1.

nallath, 2017-03-06:
This wasn't a osx only issue.

BenMarolt, 2017-07-03:
I also noticed this in **2.4** and **2.6 beta** on **Windows 7**. I've gotten into the habit of deleting the ".curaprofile.3mf" out of the filename every time I save a project to prevent the repeating extensions, but it would be nice if this could get fixed, since I forget to do the workaround sometimes and end up with messed up filenames. For a community forum post from another person who is experiencing this issue in **Cura 2.6.x**, please see <https://ultimaker.com/en/community/50109-annoying-bug-in-26x-save-project>

filipgoc, 2017-07-12:
For tracking purposes: Mac OS Sierra, Cura 2.6.2, this _issue persists._

Appesteijn, 2017-07-12:
I fixed this for MacOS in master, it should be released in 2.7.0-beta (which we hope to release in a few weeks)

closed 2017-11-17 by Ghostkeeper

BenMarolt, 2018-01-25:
Just a note why I referenced this in #3183: It seems like this issue is either still present ( or has returned ) in 3.2 Beta on Windows 7.

filipgoc, 2018-01-25:
Interesting. On Mac, it's been circumvented by simply using `.3mf` as the project extension. (This clearly has the disadvantage of not being able to tell easily 3mf meshes from 3mf projects, but it did solve this issue.)