Ultimaker/Cura #1488: multiply model option places models off build plate

TheOneTruePhradd, 2017-03-03:
Placing a model in the center of the plate, right clicking and selecting Multiply Model... will propagate models to the right off the build plate.  Other versions of Cura (Type A) correctly place the models into a grid so all are on the plate.

Ultimaker Version (adding 4 models)
[image]

Type A Version (adding 4 models)

[image]

nallath, 2017-03-06:
Cura 2 is a re-write of legacy cura (the one model A is based on). We haven't gotten to implementing this fully how it should be.

We've spent some time on checking if we could use a physics engine to handle this, which has had some promising results.  It won't be in the next (2.5) release though.

nallath, 2017-07-27:
This is in the 2.6 release.

closed 2017-07-27 by nallath