Ultimaker/Cura #1489: Generic wireless send G-Code plugin

probonopd, 2017-03-04:
I would like to see (or help making) a generic wireless send G-Code plugin. It should discover the printer using Zeroconf, and then send G-Code to the printer to stand in for this `curl` command:

```
curl -F "file=@/path/to/some.gcode" http://3d.local/print
```

I have looked at OctoPrintPlugin but it seems overkill, with JSON parsing and such. @awhiemstra suggested to look at PrinterOutputDevice (handles the printing) and OutputDevicePlugin (handles finding the device using Zeroconf).

probonopd, 2017-03-04:
Update: Started working on it, https://github.com/probonopd/WirelessPrint. Reviews, feedback, and improvements welcome.

probonopd, 2017-03-05:
[image]

Started https://github.com/probonopd/WirelessPrinting/ project, feedback, discussion, and PRs welcome.

Ghostkeeper, 2017-03-06:
Looks good. If you wish to have a plug-in included in the default Cura builds to support your project (once it's finished), let us know.

fieldOfView, 2017-03-06:
Ahem... I wish to have the OctoPrint plugin included by default.

Ghostkeeper, 2017-03-06:
I wish that too, but we made the mistake of asking that to higher management back then... This could be a first step to reverse that decision.

nallath, 2017-03-06:
Make a pull request. I will discuss it with the new product owner :) 

There is talk about having a site for easier plugin management. I'd love to have a better way to discover plugins.

fieldOfView, 2017-03-06:
This is getting offtopic, but it would be nice if there were a plugin downloader/updater in Cura. I think that would be better in a way than shipping and enabling everything by default; the machine actions section would quickly get very crowded with "Connect with OctoPrint" "Connect via USB" "Connect WirelessPrint" etc buttons on all USB-enabled printers.

awhiemstra, 2017-03-06:
Well, yes. However, this needs both a web component and a Cura component and is not a trivial amount of work. It also would be nice if we do not need to re-invent the wheel and would be able to reuse an existing implementation. Unfortunately, I only know KDE's KNewStuff as a library to provide this functionality and I do not know how usable that is for Cura.

probonopd, 2017-03-06:
> the machine actions section would quickly get very crowded with "Connect with OctoPrint" "Connect via USB" "Connect WirelessPrint" etc buttons on all USB-enabled printers

I had opened an issue about this very question on https://github.com/probonopd/WirelessPrinting/issues/7:

> Since the WirelessPrinting plugin can be used with _every printer_ that accepts G-Code, it would be good if Cura had an option to hook such plugins into every printer using a different mechanism that doesn't involve spamming all machine definition files.

nallath, 2017-07-27:
Octoprint plugin is now added. Whoo.

closed 2017-07-27 by nallath

probonopd, 2017-08-13:
Changed https://github.com/probonopd/WirelessPrinting to use a subset of the OctoPrint protocol so that OctoPrintPlugin can be used with it. Whoo!