Ultimaker/Cura #1490: Linux: Does not load plugins from Configuration Folder

probonopd, 2017-03-05:
* Cura 2.4 AppImage on Ubuntu 16.04
* Help -> Show Configuration Folder
* Manually create a directory `plugins` there (there is none by default)
* Put a plugin into it
* Restart Cura

The plugin is not found.

This is my directory structure:

```
/home/me/.config/cura/
/home/me/.config/cura/cura.cfg
/home/me/.config/cura/plugins
/home/me/.config/cura/plugins/WirelessPrinting/
/home/me/.config/cura/plugins/WirelessPrinting/...
```

I also tried to put the `WirelessPrinting` directory directly into the `/home/me/.config/cura/` directory but that also does not work.

It would be nice if there was a `plugins` directory by default in the Configuration Folder, and if plugins would actually be loaded from there.

fieldOfView, 2017-03-05:
Could you check if there is a ~/.local/share/cura folder which may contain a plugins folder?

probonopd, 2017-03-05:
Sure enough, I do have

```
me@host:~$ find ~/.local/share/cura/
/home/me/.local/share/cura/
/home/me/.local/share/cura/machine_instances
/home/me/.local/share/cura/extruders
/home/me/.local/share/cura/user
/home/me/.local/share/cura/materials
/home/me/.local/share/cura/variants
/home/me/.local/share/cura/quality
/home/me/.local/share/cura/cura.log
/home/me/.local/share/cura/plugins
```

fieldOfView, 2017-03-05:
That's the plugins folder you're supposed to use.

probonopd, 2017-03-05:
Thanks, updating documentation accordingly.

probonopd, 2017-03-05:
Help -> Show Configuration Folder should open `$HOME/.local/share/cura/`.

awhiemstra, 2017-03-06:
Strictly speaking, the entry is correct since ~/.config/cura is the 'configuration' folder. It is just that on Linux according to XDG specs the configuration and local data folders are separated and we have no real way of distinguishing the two.

That said, the menu entry should open the folder with the logs and other stuff you usually want to actually look at and so should open the local data folder on Linux instead of the config folder.

Ghostkeeper, 2017-03-06:
Or both? Or would that be annoying.

probonopd, 2018-05-16:
@alekseisasin I haven't re-tested it so far since I didn't assume this aspect had been worked on so far. Can't test right now since not on a Linux system atm.

alekseisasin, 2018-05-17:
I only tested in master build, the plugin structure will be changed and I think the new solution will be available in next release.  
(master build)For the testing I added manually plugin to the 
/home/me/.local/share/cura/plugins
Adjusted packages.json /home/me/.local/share/cura/
And it will work.

probonopd, 2018-05-17:
@alekseisasin what does Help -> Show Configuration Folder open on your system?

alekseisasin, 2018-05-17:
I think this is the most simplest way to add manually plugin. All these steps I did in Cura 3.3.1
1- Archive your plugin folder. I used zip (like "yourplugin.zip")
2- Change zipped file extension to .curaplugin (like "yourplugin.curaplugin")
3- Drag and drop the plugin into cura.
After it you will see the message whether plugin was installed or not. To finish please restart the Cura.

commit b33bbde2 (2018-06-20): Also open data folder when opening Resources folder

So now it opens two folders. It could be slightly more annoying to some but usually it's much more useful.

Fixes #1490.
closed 2018-06-20 by Ghostkeeper