Ultimaker/Cura #1493: Feature request: sequential printing without completing objects

Patola, 2017-03-05:
This is a feature present in Simplify3D, described here: https://www.simplify3d.com/support/articles/multi-part-printing/ (Multiple Process, Sequential Printing Mode)
The main advantage is that it saves a lot of travel moves and prints much faster. But it also improves on reliability.

### Version
I am currently using Cura 2.4.99-master from thopiekar's repository 

### Operating system type + version
Ubuntu 16.04 64 bits

### Behavior
When using sequential printing, if objects are taller than the heigh clearance Cura simply refuses to slice. But it should not be this way. It could behave like Simplify3D: since we are limited by this height, do a sequential printing for all objects up to a height of _height clearance_, then start over again at _height clearance+1_ and do it sequentially again, then again at _height clearance × 2 + 1_ and so on.

_Is this a new feature request?_
YES

Note 1: in the picture, gantry height is at 20mm.
Note 2: I found issue #1431 which deals with an "enhancement" for 2.5mm. I could not figure out whether the original poster was describing the same feature I am describing, I did not understand his/her wording very well. If that is the case, feel free to close this issue.

[image]

Ghostkeeper, 2017-03-06:
That is indeed how I interpreted #1493. I'll close this one and add a bit of information to the other one.

ThomasGreska, 2017-03-06:
An it is indeed what I was thinking off when writing #1431.
Let's wait another six month, maybe time solves the problem :-)

closed 2017-03-07 by Ghostkeeper