Ultimaker/Cura #1496: Very confusing cross references between definitions and qualities

ubitux, 2017-03-06:
I'm trying to add a new printer, but the I'm unable to get the `quality` files correctly linked to their printer definition: apparently, a match between `definition` in the quality cfg file and `id` in the definition json file is not enough to make the link.

I need to understand how what values are used, because currently I have no idea which elements in this list influence the link:

- `id` value in the definition file
- `name` value in the definition file (`.lower().replace(' ','_')`?)
- `overrides.machine_name.default_value` value in the definition file (yet another name?)
- `metadata.manufacturer` value in the definition file
- filename of the definition file
- directory of the quality config files
- filename of the quality config files
- `general.definition` value in the quality files
- `metadata.material` value in the quality file (maybe it needs the name in it somehow?)

I've tried many combination, but I'm yet to see any profile show up when I set `has_machine_quality` in the definition file. Maybe there is no bug, but at least there is an issue wrt documentation.

nallath, 2017-03-07:
The name of the file is used as the id (eg; fdmprinter.def.json gives it the id fdmprinter).

When messing around with new profiles I can recommend using the god-mode plugin for Cura. It gives you a whole lot more debug information. Also take a good look at the log files.

We know it's a bit annoying how all of this works now, but the system has grown a bit complex over the past few months.

Ghostkeeper, 2017-03-07:
Indeed it is the filename of the definition file (5th point in your list). To fix this, we should remove some of the things in the list.

1. The `id` value in the definition file is completely superfluous now. We should remove it.
2. The `name` value in the definition file is the user-visible name in the Cura GUI.
3. The `overrides.machine_name.default_value` duplicates the `name` metadata. If it is removed however, we should find a different way to communicate the printer name to CuraEngine. Perhaps the only good solution is to remove 2. and use this value in the GUI instead. The problem with removing 2. is that this is implemented in Uranium and `machine_name` is Cura-specific.
4. `metadata.manufacturer` is a completely unrelated metadata property and should not be removed.
5. The file name of the definition file obviously can't be removed. It's also used basically everywhere.
6. The directory of the quality profiles is not used in the code. We can remove it but it helps with overview.
7. The file name of the quality profiles is the same as 5. Also used for their ID. We can't remove the file name and it's not duplicate.
8. `general/definition` refers to 5. It is used to link to a definition, which is duplicate with 6, but I'm not sure we want to remove 6.
9. Same as 8.

In the end, the only ones we want/can remove are 1 and maybe 2.

ubitux, 2017-03-07:
@nallath as recommended on IRC, I used the god-mode plugin (with a suggested patch to make it work with git/master). I was able to fix the various traceback thanks to it but it didn't help with the issue.

Issue being: the material and profile are still decorrelated in Cura (I'm supposed to only have a limited number of profile for a given material). More worrying, picking the material has no impact on the material diameter (displayed in the Print Setup), even though going in the Material Manager window displays the correct diameter. So I'm assuming there is still some kind of reference(s) broken somewhere.

@Ghostkeeper so if I understand correctly, the file names matter. So this tree is supposed to work:

```
output
├── definitions
│   └── dagoma_discoeasy200.def.json
├── materials
│   ├── chromatik_pla.xml.fdm_material
│   ├── fiberlogy_hd_pla.xml.fdm_material
│   ├── filo3d_pla_green.xml.fdm_material
│   ├── filo3d_pla_red.xml.fdm_material
│   ├── filo3d_pla.xml.fdm_material
│   ├── octofiber_pla.xml.fdm_material
│   ├── polyflex_pla.xml.fdm_material
│   ├── polymax_pla.xml.fdm_material
│   ├── polyplus_pla.xml.fdm_material
│   └── polywood_pla.xml.fdm_material
├── meshes
│   └── dagoma_discoeasy200_platform.stl
└── quality
    └── dagoma_discoeasy200
        ├── dagoma_discoeasy200_chromatik_pla_fast.inst.cfg
        ├── dagoma_discoeasy200_chromatik_pla_high.inst.cfg
        ├── dagoma_discoeasy200_chromatik_pla_normal.inst.cfg
        ├── dagoma_discoeasy200_fiberlogy_hd_pla_fast.inst.cfg
        ├── dagoma_discoeasy200_fiberlogy_hd_pla_high.inst.cfg
        ├── dagoma_discoeasy200_fiberlogy_hd_pla_normal.inst.cfg
        ├── dagoma_discoeasy200_filo3d_pla_fast.inst.cfg
        ├── dagoma_discoeasy200_filo3d_pla_green_fast.inst.cfg
        ├── dagoma_discoeasy200_filo3d_pla_green_high.inst.cfg
        ├── dagoma_discoeasy200_filo3d_pla_green_normal.inst.cfg
        ├── dagoma_discoeasy200_filo3d_pla_high.inst.cfg
        ├── dagoma_discoeasy200_filo3d_pla_normal.inst.cfg
        ├── dagoma_discoeasy200_filo3d_pla_red_fast.inst.cfg
        ├── dagoma_discoeasy200_filo3d_pla_red_high.inst.cfg
        ├── dagoma_discoeasy200_filo3d_pla_red_normal.inst.cfg
        ├── dagoma_discoeasy200_octofiber_pla_fast.inst.cfg
        ├── dagoma_discoeasy200_octofiber_pla_high.inst.cfg
        ├── dagoma_discoeasy200_octofiber_pla_normal.inst.cfg
        ├── dagoma_discoeasy200_polyflex_pla_flexible.inst.cfg
        ├── dagoma_discoeasy200_polymax_pla_fast.inst.cfg
        ├── dagoma_discoeasy200_polymax_pla_high.inst.cfg
        ├── dagoma_discoeasy200_polymax_pla_normal.inst.cfg
        ├── dagoma_discoeasy200_polyplus_pla_fast.inst.cfg
        ├── dagoma_discoeasy200_polyplus_pla_high.inst.cfg
        ├── dagoma_discoeasy200_polyplus_pla_normal.inst.cfg
        └── dagoma_discoeasy200_polywood_pla_wood.inst.cfg
```

```
%  head -n 10 output/quality/dagoma_discoeasy200/dagoma_discoeasy200_chromatik_pla_fast.inst.cfg
[general]
version = 2
name = Fast (0.2mm)
definition = dagoma_discoeasy200

[metadata]
type = quality
quality_type = fast
material = chromatik_pla

```

What else should I check? Cura logger only raises tracebacks about `Savitar` and `zeroconf` not being found. I'm using git/master in the 4 main repositories (Cura, CuraEngine, Uranium, and libArcus) and clean/rebuild everything (install + cache + local data) when doing tests.

Ghostkeeper, 2017-03-08:
The material diameter thing is a different problem (reported more often). There's basically two material diameter settings and we should unite them.

That file tree looks like it's supposed to work all right. Note that most of the file names there don't matter to Cura (as long as they are linked correctly in other files), but are only to organise the files for better maintainability. Is it not working? Does your log say anything about these profiles?

ubitux, 2017-03-08:
How am I supposed to check which diameter is going to be used then?

It doesn't seem to be working because picking a normal PLA such as Chromatik PLA gives me the possibility to pick any profile such as Wood and Flexible. And similarly, picking a flexible-only material allows me to pick any profile.

Here is my current log:

```
UM.Preferences.getValue [81]: Tried to get the value of non-existing setting general/language.
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [103]: Trying OpenGL context 4.1...
UM.View.GL.OpenGLContext.detectBestOpenGLVersion [113]: Yay, we got at least OpenGL 4.1 core: 4.5 Core profile
UM.Qt.QtApplication.__init__ [71]: Detected most suitable OpenGL context version: 4.1 Core profile
2017-03-08 10:58:21,179 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin ConsoleLogger
2017-03-08 10:58:21,184 - INFO - CuraEngineBackend.CuraEngineBackend.__init__ [64]: Found CuraEngine at: /home/ux/.local/bin/CuraEngine
2017-03-08 10:58:21,184 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin CuraEngineBackend
2017-03-08 10:58:21,186 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin CameraTool
2017-03-08 10:58:21,187 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin MirrorTool
2017-03-08 10:58:21,188 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin SelectionTool
2017-03-08 10:58:21,190 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin RotateTool
2017-03-08 10:58:21,191 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin TranslateTool
2017-03-08 10:58:21,193 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin ScaleTool
2017-03-08 10:58:21,194 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin FileLogger
2017-03-08 10:58:21,194 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin SimpleView
2017-03-08 10:58:21,195 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin WireframeView
2017-03-08 10:58:21,195 - WARNING - UM.PluginRegistry.loadPlugin [74]: Plugin ConsoleLogger was already loaded
2017-03-08 10:58:21,197 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin LocalFileOutputDevice
2017-03-08 10:58:21,199 - WARNING - STLReader.STLReader.<module> [26]: Could not find numpy-stl, falling back to slower code.
2017-03-08 10:58:21,200 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin STLReader
2017-03-08 10:58:21,201 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin STLWriter
2017-03-08 10:58:21,201 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin OBJReader
2017-03-08 10:58:21,202 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin OBJWriter
2017-03-08 10:58:21,229 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin UpdateChecker
2017-03-08 10:58:21,235 - INFO - UpdateChecker.UpdateChecker.run [40]: Checking for new version of cura
2017-03-08 10:58:21,238 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin GodMode
2017-03-08 10:58:21,238 - WARNING - UM.PluginRegistry.loadPlugin [74]: Plugin CuraEngineBackend was already loaded
2017-03-08 10:58:21,240 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin CuraProfileReader
2017-03-08 10:58:21,241 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin ImageReader
2017-03-08 10:58:21,242 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin LegacyProfileReader
2017-03-08 10:58:21,245 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin X3DReader
2017-03-08 10:58:21,246 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin CuraProfileWriter
2017-03-08 10:58:21,247 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin SliceInfoPlugin
2017-03-08 10:58:21,249 - INFO - UM.PluginRegistry.loadPlugin [102]: Loaded plugin MachineSettingsAction
2017-03-08 10:58:21,250 - ERROR - UM.Logger.logException [74]: Exception: Import error loading module 3MFWriter
2017-03-08 10:58:21,251 - ERROR - UM.Logger.logException [78]: Traceback (most recent call last):
2017-03-08 10:58:21,251 - ERROR - UM.Logger.logException [78]:   File "/home/ux/.local/lib/python3.6/site-packages/UM/PluginRegistry.py", line 293, in _findPlugin
2017-03-08 10:58:21,251 - ERROR - UM.Logger.logException [78]:     module = imp.load_module(plugin_id, file, path, desc)
2017-03-08 10:58:21,251 - ERROR - UM.Logger.logException [78]:   File "/usr/lib/python3.6/imp.py", line 244, in load_module
2017-03-08 10:58:21,251 - ERROR - UM.Logger.logException [78]:     return load_package(name, filename)
2017-03-08 10:58:21,251 - ERROR - UM.Logger.logException [78]:   File "/usr/lib/python3.6/imp.py", line 216, in load_package
2017-03-08 10:58:21,251 - ERROR - UM.Logger.logException [78]:     return _load(spec)
2017-03-08 10:58:21,252 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap>", line 675, in _load
2017-03-08 10:58:21,252 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap>", line 655, in _load_unlocked
2017-03-08 10:58:21,252 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap_external>", line 678, in exec_module
2017-03-08 10:58:21,252 - ERROR - UM.Logger.logException [78]:   File "<frozen importlib._bootstrap>", line 205, in _call_with_frames_removed
2017-03-08 10:58:21,252 - ERROR - UM.Logger.logException [78]:   File "/home/ux/.local/lib/cura/plugins/3MFWriter/__init__.py", line 6, in <module>
2017-03-08 10:58:21,252 - ERROR - UM.Logger.logException [78]:     from . import ThreeMFWriter
[... cut ...]
2017-03-08 10:58:30,963 - DEBUG - USBPrinting.USBPrinterOutputDevice._connect [348]: Correct response for auto-baudrate detection received.
2017-03-08 10:58:30,964 - INFO - USBPrinting.USBPrinterOutputDevice._connect [355]: Established printer connection on port /dev/ttyUSB0
2017-03-08 10:58:30,964 - INFO - USBPrinting.USBPrinterOutputDevice._listen [477]: Printer connection listen thread started for /dev/ttyUSB0
2017-03-08 10:58:33,990 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
```

Ghostkeeper, 2017-03-08:
> It doesn't seem to be working because picking a normal PLA such as Chromatik PLA gives me the possibility to pick any profile such as Wood and Flexible. And similarly, picking a flexible-only material allows me to pick any profile.

So that was the problem! I thought it was crashing for you or not adding the definition or something.

It seems to be an issue of documentation then, because all your material profiles have the PLA material type (if they are named accurately). Profiles are filtered on the material type, not on the individual material. If something like woodfill or flexible material needs different quality profiles, then it should have a different material type.

If profiles were linked to materials directly rather than material types, then every time a user duplicates a material they'd have to create wholly new profiles for it, even if they just want to adjust the temperature or something.

ubitux, 2017-03-08:
So wait, currently I have 3 cases of profiles:
- wood generated for the polywood material
- flexible generated for the polyflex material
- fast/high/normal generated for each of the remaining PLA materials (chromatik, fiberlogy, filo3d, ...)

What happens if I pick, let's say, the not supposed to be available "fast" profile in the menu when I have the polyflex material? Which settings is it going to use?

Ghostkeeper, 2017-03-08:
If polyflex has its own material type and `dagoma_discoeasy200_polyflex_pla_flexible.inst.cfg` is the only available profile for that material type, then you should only be seeing that quality profile in the drop-down menu, no "fast" profile.

Note that Polyflex is not PLA, by the way. Its major component is TPU.

ubitux, 2017-03-08:
I'm seeing the 5 profiles for every material I'm picking: fast, high, normal, wood and flexible, be it PLA, flexible or wood. That's what I'm trying to fix.

About Polyflex not being PLA, I'm only porting data from a fork of Cura Legacy, and it says "PLA". Not saying it's correct :)

I'll change PLA to TPU when this is actually working.

ubitux, 2017-03-09:
(Apparently Github dropped my post or I'm going crazy, so I'm reposting again).

But instead of a long text, here is a gif to clarify the behavior:

[image]

Whatever the material I always get all the profiles/quality. But if I manually put garbage into the `metadata.material` of the quality config files, that profile list gets empty, which is the expected behavior but also confirms the link is currently "working".

Still, I'm not supposed to get wood and flexible with the random PLA, and should not get anything else than wood and flexible with the respective polywood and polyflex:

```
quality/dagoma_discoeasy200/dagoma_discoeasy200_chromatik_pla_fast.inst.cfg:quality_type = fast
quality/dagoma_discoeasy200/dagoma_discoeasy200_chromatik_pla_high.inst.cfg:quality_type = high
quality/dagoma_discoeasy200/dagoma_discoeasy200_chromatik_pla_normal.inst.cfg:quality_type = normal
quality/dagoma_discoeasy200/dagoma_discoeasy200_fiberlogy_hd_pla_fast.inst.cfg:quality_type = fast
quality/dagoma_discoeasy200/dagoma_discoeasy200_fiberlogy_hd_pla_high.inst.cfg:quality_type = high
quality/dagoma_discoeasy200/dagoma_discoeasy200_fiberlogy_hd_pla_normal.inst.cfg:quality_type = normal
quality/dagoma_discoeasy200/dagoma_discoeasy200_filo3d_pla_fast.inst.cfg:quality_type = fast
quality/dagoma_discoeasy200/dagoma_discoeasy200_filo3d_pla_green_fast.inst.cfg:quality_type = fast
quality/dagoma_discoeasy200/dagoma_discoeasy200_filo3d_pla_green_high.inst.cfg:quality_type = high
quality/dagoma_discoeasy200/dagoma_discoeasy200_filo3d_pla_green_normal.inst.cfg:quality_type = normal
quality/dagoma_discoeasy200/dagoma_discoeasy200_filo3d_pla_high.inst.cfg:quality_type = high
quality/dagoma_discoeasy200/dagoma_discoeasy200_filo3d_pla_normal.inst.cfg:quality_type = normal
quality/dagoma_discoeasy200/dagoma_discoeasy200_filo3d_pla_red_fast.inst.cfg:quality_type = fast
quality/dagoma_discoeasy200/dagoma_discoeasy200_filo3d_pla_red_high.inst.cfg:quality_type = high
quality/dagoma_discoeasy200/dagoma_discoeasy200_filo3d_pla_red_normal.inst.cfg:quality_type = normal
quality/dagoma_discoeasy200/dagoma_discoeasy200_octofiber_pla_fast.inst.cfg:quality_type = fast
quality/dagoma_discoeasy200/dagoma_discoeasy200_octofiber_pla_high.inst.cfg:quality_type = high
quality/dagoma_discoeasy200/dagoma_discoeasy200_octofiber_pla_normal.inst.cfg:quality_type = normal
quality/dagoma_discoeasy200/dagoma_discoeasy200_polyflex_pla_flexible.inst.cfg:quality_type = flexible
quality/dagoma_discoeasy200/dagoma_discoeasy200_polymax_pla_fast.inst.cfg:quality_type = fast
quality/dagoma_discoeasy200/dagoma_discoeasy200_polymax_pla_high.inst.cfg:quality_type = high
quality/dagoma_discoeasy200/dagoma_discoeasy200_polymax_pla_normal.inst.cfg:quality_type = normal
quality/dagoma_discoeasy200/dagoma_discoeasy200_polyplus_pla_fast.inst.cfg:quality_type = fast
quality/dagoma_discoeasy200/dagoma_discoeasy200_polyplus_pla_high.inst.cfg:quality_type = high
quality/dagoma_discoeasy200/dagoma_discoeasy200_polyplus_pla_normal.inst.cfg:quality_type = normal
quality/dagoma_discoeasy200/dagoma_discoeasy200_polywood_pla_wood.inst.cfg:quality_type = wood
```

```
quality/dagoma_discoeasy200/dagoma_discoeasy200_chromatik_pla_fast.inst.cfg:material = chromatik_pla
quality/dagoma_discoeasy200/dagoma_discoeasy200_chromatik_pla_high.inst.cfg:material = chromatik_pla
quality/dagoma_discoeasy200/dagoma_discoeasy200_chromatik_pla_normal.inst.cfg:material = chromatik_pla
quality/dagoma_discoeasy200/dagoma_discoeasy200_fiberlogy_hd_pla_fast.inst.cfg:material = fiberlogy_hd_pla
quality/dagoma_discoeasy200/dagoma_discoeasy200_fiberlogy_hd_pla_high.inst.cfg:material = fiberlogy_hd_pla
quality/dagoma_discoeasy200/dagoma_discoeasy200_fiberlogy_hd_pla_normal.inst.cfg:material = fiberlogy_hd_pla
quality/dagoma_discoeasy200/dagoma_discoeasy200_filo3d_pla_fast.inst.cfg:material = filo3d_pla
quality/dagoma_discoeasy200/dagoma_discoeasy200_filo3d_pla_green_fast.inst.cfg:material = filo3d_pla_green
quality/dagoma_discoeasy200/dagoma_discoeasy200_filo3d_pla_green_high.inst.cfg:material = filo3d_pla_green
quality/dagoma_discoeasy200/dagoma_discoeasy200_filo3d_pla_green_normal.inst.cfg:material = filo3d_pla_green
quality/dagoma_discoeasy200/dagoma_discoeasy200_filo3d_pla_high.inst.cfg:material = filo3d_pla
quality/dagoma_discoeasy200/dagoma_discoeasy200_filo3d_pla_normal.inst.cfg:material = filo3d_pla
quality/dagoma_discoeasy200/dagoma_discoeasy200_filo3d_pla_red_fast.inst.cfg:material = filo3d_pla_red
quality/dagoma_discoeasy200/dagoma_discoeasy200_filo3d_pla_red_high.inst.cfg:material = filo3d_pla_red
quality/dagoma_discoeasy200/dagoma_discoeasy200_filo3d_pla_red_normal.inst.cfg:material = filo3d_pla_red
quality/dagoma_discoeasy200/dagoma_discoeasy200_octofiber_pla_fast.inst.cfg:material = octofiber_pla
quality/dagoma_discoeasy200/dagoma_discoeasy200_octofiber_pla_high.inst.cfg:material = octofiber_pla
quality/dagoma_discoeasy200/dagoma_discoeasy200_octofiber_pla_normal.inst.cfg:material = octofiber_pla
quality/dagoma_discoeasy200/dagoma_discoeasy200_polyflex_pla_flexible.inst.cfg:material = polyflex_pla
quality/dagoma_discoeasy200/dagoma_discoeasy200_polymax_pla_fast.inst.cfg:material = polymax_pla
quality/dagoma_discoeasy200/dagoma_discoeasy200_polymax_pla_high.inst.cfg:material = polymax_pla
quality/dagoma_discoeasy200/dagoma_discoeasy200_polymax_pla_normal.inst.cfg:material = polymax_pla
quality/dagoma_discoeasy200/dagoma_discoeasy200_polyplus_pla_fast.inst.cfg:material = polyplus_pla
quality/dagoma_discoeasy200/dagoma_discoeasy200_polyplus_pla_high.inst.cfg:material = polyplus_pla
quality/dagoma_discoeasy200/dagoma_discoeasy200_polyplus_pla_normal.inst.cfg:material = polyplus_pla
quality/dagoma_discoeasy200/dagoma_discoeasy200_polywood_pla_wood.inst.cfg:material = polywood_pla
```

Ghostkeeper, 2017-03-10:
It is indeed confusing. What happens when you change this line?
https://github.com/DelphinPETER/Dagoma-Settings-For-Cura/blob/Cura_2.4/resources/materials/polywood_pla.xml.fdm_material#L9

To something like `<material>Woodfill</material>`.
That is the "material type" I was talking about.

ubitux, 2017-03-10:
This seems to work: the "wood" profile disappears when picking PLA materials, and is only present with PolyWood material.

I understand we may want to change that key, but even if we do that, I can't be sure that picking fast/high/normal in the other PLA will actually apply the expected profile...

Note: current repository reference is located at https://github.com/ubitux/cura-dagoma/tree/master/output
Note 2: the listed materials are already upstream at https://github.com/Ultimaker/fdm_materials/

Ghostkeeper, 2017-03-13:
> I understand we may want to change that key, but even if we do that, I can't be sure that picking fast/high/normal in the other PLA will actually apply the expected profile...

Cura makes sure that when a user selects a quality profile on one extruder, all other extruders always have a quality profile with the same quality type at the other extruder. The quality types we have are normal, high, fast and draft. So if you'd select Normal quality for the extruder doing Filo3D PLA while the other extruder has Octofiber, it would select your `dagoma_discoeasy200_filo3d_pla_normal.inst.cfg` file on one extruder and the `dagoma_discoeasy_octofiber_pla_normal.inst.cfg` in the other, since both files have quality type `normal`. Switching one of them to High would switch the other too.

Also, when a user makes a custom quality profile (called "quality-changes" internally), they get the same quality type as the profile it was derived from.

ubitux, 2017-03-13:
OK. So in the end, do we agree that there is one bug when only 1 quality is supposed to be available but many of them ends up being available as well? Like, in the current case of PolyWood: what happens when I pick `Standard`? There is no such thing as `dagoma_discoeasy200_polywood_pla_normal.inst.cfg`, only `dagoma_discoeasy200_polywood_pla_wood.inst.cfg`

Ghostkeeper, 2017-03-14:
If you've got Polywood selected as material and it has its own material type, then "standard" should not be in the list to select.

ubitux, 2017-03-14:
Why "and has its own material type"?

We are defining specific configurations for specific materials, not material types! Are you saying that quality profiles can not be specific to a material if they are of the same type?

I'm sorry, it looks like we have a huge misunderstanding somewhere, but I'm starting to believe it's not just a bug but a design issue...

Ghostkeeper, 2017-03-15:
It is a design issue. The code that handles the selecting of materials and quality profiles is extremely involved: https://github.com/Ultimaker/Cura/blob/master/cura/Settings/MachineManager.py

It might be a good idea at this point to use the God Mode plug-in https://github.com/sedwards2009/cura-god-mode-plugin
It'll show exactly which profiles are loaded so there is no confusion due to profiles having the same display name.

ubitux, 2017-03-15:
So the proper workaround is to change the material type of PolyWood and PolyFlex to something arbitrary different than all the other materials available for a given printer, otherwise I end up with undefined behaviour (because unavailable profiles becomes visible and we don't know which one applies)?

Is there a plan or issue opened about that design problem?

Ghostkeeper, 2017-03-16:
It's not undefined behaviour, it's just very complex. That is a design issue.

We have a refactor planned for the way that resolve functions work in 2.6, but that is not really related. We also want to change the way that material profiles work one day, which is related, but that is not planned for any specific release yet.

ianpaschal, 2018-02-27:
@Ghostkeeper Is this still and issue? I'm also not super experienced at making definitions or qualities.

closed 2018-03-02 by ChrisTerBeke