Ultimaker/Cura #1498: Too many branches

fieldOfView, 2017-03-07:
The Cura repository currently has 46 branches, many of which have already been merged or will never be merged. This includes a couple of mine, but I am no longer allowed to remove these branches.
* fix_unsupported_materials_warning
* fix_um2p_profilenames
* feature_black

Other branches that have already been merged: smartavionics-mb-anchor-skin-in-infill, feature_auto_slice, feature_temp_disable, python_type_hinting, feature_preheat_bed, single_instance, layer_view3. And those are just the "active" branches.

Time for some spring cleaning?

nallath, 2017-03-07:
I've killed all the branches you mentioned. It's a good start at least. I also noticed that i have a ton of local branches that need purging. Better get started with that as well.

nickthetait, 2017-03-08:
I agree, quite a kerfuffle at the moment :stuck_out_tongue:  Might add a step to the end of your release process to prune the old ones.

BagelOrb, 2017-03-08:
I agree.
My usual cycle is 
```
git log some_branch ^master  --pretty=oneline
#if result is no commits then:
git push origin :some_branch
git branch -D some_branch
```

I think it's better not to automate this, because you might want to keep some branches a bit longer sometimes, even when they are already merged.

BagelOrb, 2017-03-08:
Also: there's too many open issues which can be closed. Keeping github clean means that contributors have clear information and we have clear information on what the community wants or the bugs they ahave found which haven't been solved yet.

fieldOfView, 2017-04-02:
These branches are from PRs that have been merged:
```
python_type_hinting
recommended_sidebar
```

The following branches have no commits ahead of master (so they were also merged):
```
feature_omp_writeGcode
feature_z_seam_location
feature_recursive_cubic
feature_alternate_carving
feature_bed_adhesion_none
feature_precool
feature_fan_speed_0
feature_extruder_temp_layer_0
feature_wall_wipe
feature_anti_support
feature_infill_support
feature_retract_at_layer_change
feature_bed_temp_layer_0
feature_concentric_3d_infill
feature_better_warning_formulas
feature_resolvement_functions
bugfix_global_inherits_stack
```
Note they are (allmost?) all @BagelOrb's branches, so somewhere there's a workflow merging his stuff that is not working out.

Finally, these branches are from PRs that have been retracted:
```
robo_c2
thopiekar-unity-workaround
bugfix_legacy_importer
```

Together that's 22 unnecessary branches.

nallath, 2017-04-03:
@BagelOrb usually doesn't make pull requests, so then there is no reminder to remove old branches. I think we should indeed have a workflow change (eg; always use a pull request)

awhiemstra, 2017-04-03:
I have gone through the branches and deleted a bunch of them. Branch count is now down to 17 with most of these being active, release, or PR related branches.

BagelOrb, 2018-04-09:
@fieldOfView You still got privileges?! O_O

Uhmmm @Ghostkeeper .....?

BagelOrb, 2018-04-09:
oh wait nvm. Everybody can open tickets of course! :P

fieldOfView, 2018-04-09:
It is that time again... The following branches have no commits ahead of master:

feature_try_line_widths
CURA-5135_fancy_fix 
feature_gzgcode_writer
CURA-4800_test_opengl
CURA-4916_infill_sidebar_update
CURA-4968_support_eraser_on_platform

Ghostkeeper, 2018-05-01:
I cleaned those up again.

closed 2018-05-01 by Ghostkeeper