Ultimaker/Cura #1503: Feature request: build plate adhesion type / setting per model

Dim3nsioneer, 2017-03-07:
Please add the, maybe restricted, possibility to set the build plate adhesion typ per model. This would help in situations where you print a collection of parts at the same time of which some parts require a brim while for others a brim is not very nice to have due to maybe small details in the first layer.

BagelOrb, 2017-03-08:
Having different adhesion types is rather difficult, because the adhesion of different models can interfere.
We could however, place the brim of an object further away from the object, so that it doesn't touch the object.

A more difficult approach would be to only generate the brim around some models, and then cut the lines of the brim at the places where those brim lines overlap with the other model.

Ghostkeeper, 2017-03-08:
I think we can find a way to subtract the polygons of the brim and skirt or something. Requires some thought.

The greater problem is with raft. If there's extra layers below the model then just the single model with raft has to be moved upwards.

BagelOrb, 2017-03-08:
The raft thing is probably not going to happen, but we could make some exception-setting for Print This Model Without Brim

Dim3nsioneer, 2017-03-09:
Those ideas sound quite promising to me! 👍

Ghostkeeper, 2017-06-05:
Our product manager just closed this because it was inactive for 3 months.

Dim3nsioneer, 2017-06-06:
What does that mean for the feature request? Is it therefore denied?

nallath, 2017-06-06:
it pretty much means: We're getting 100 "worth" of feature requests, with about 20 "worth" of development time.
In order to be able to keep track of this big pile up of requests, our product manager takes the nuclear option every so often.

UltiArjan, 2017-06-06:
IMHO this impacts something far more important, I'm hoping since last year October to get the option of Brim ONLY on for the wipe tower, and not for the  print item.....

nickthetait, 2017-06-07:
UltiArjan is that because the wipe tower is in some way deficient in how it is attached to the bed or more specifically because this feature is needed?

nallath, 2017-06-08:
Setting something per model is quite different from setting something for the wipetower. The wipe tower isn't a model. 
So if we would have added this feature, the wipe tower still wouldn't have a brim.

UltiArjan, 2017-06-08:
Ok, clear, it's been discussed before, but I'll start a new feature request for brim on the tower

Ghostkeeper, 2017-06-10:
We do have a different feature open for a while that says "Allow brim for prime tower only".

allyourcode, 2018-10-27:
+1 for this feature. I want to experiment with different types. Without this feature, my experiment(s) have to be done as separate prints, instead of just one. Sad!

Ghostkeeper, 2019-01-18:
Brim for prime tower is now implemented and under review: https://github.com/Ultimaker/CuraEngine/pull/933