Ultimaker/Cura #1507: Feature request: add travel_fspeed (GCode)

ubitux, 2017-03-07:
I have a printer that needs in its start/end GCode access to the `{travel_speed} * 60` because that's how the GCode is supposed to interpret that speed in mm/minute, not mm/second (unit of `{travel_speed}`). The GCode doesn't seem to support expression such as `G1 ... F50*60`, at least not with the printer I'm trying to use.

Ghostkeeper, 2017-03-08:
Are you creating a complete printer definition (.def.json) or are you using the Custom FDM Printer functionality?

If you're editing a .def.json file, you can do this by overriding the `"value"` property rather than `"default_value"`. The `"value"` property is interpreted with Python evaluation, so it must be Python code in there that adds some strings together. You'll get something like:

    "machine_start_gcode": {
        "value": "'M190 S60 #Heat up buildplate\\nG0 X10 Y10 F' + str(speed_travel * 60) + '#Move to left bottom corner\\nG1 E8 #Prime nozzle.'"
    }

Note the double quotes at the beginning and end, since the double quotes are required by the JSON format and then inside the quotes it defines a string in Python. Also note the double escaping of `\\n`, since both JSON and Python will unescape.

If you're using a Custom FDM Printer with some start g-code, you can only do this by editing the actual file of your printer. Look in your configuration folder (in Cura's application menu go to Help -> Show Configuration Folder). Then find the variants folder and the file in there with the name of the printer that you're modifying. In there you can add a formula to a setting by making it start with `=`. You'd get something like this:

    [values]
    machine_start_gcode = ='M190 S60 #Heat up buildplate\nG0 X10 Y10 F' + str(speed_travel * 60) + '#Move to left bottom corner\nG1 E8 #Prime nozzle.'

ubitux, 2017-03-08:
Yes I'm writing a complete printer definition.

So this is supposed to work?

```
        "machine_start_gcode": {
            "value": "'G90   ;absolute positioning\\nM106 S250 ;fan on for the palpeur\\nG28 X Y\\nG1 X50\\nM109 S180\\nG28\\nM104 S{material_print_temperature}\\nM107 ;start with the fan off\\nG1 X100 Y20 F3000\\nG1 Z0.5\\nM109 S{material_print_temperature}\\nM82 ;set extruder to absolute mode\\nG92 E0 ;zero the extruded length\\nG1 F200 E10 ;extrude 10mm of feed stock\\nG92 E0 ;zero the extruded length again\\nG1 Z3\\nG1 F' + str(speed_travel * 60) + ''"
        },
        "machine_end_gcode": {
            "value": "'M104 S0\\nM106 S255   ;start fan full power\\nM140 S0   ;heated bed heater off (if you have it)\\nG91   ;relative positioning\\nG1 E-1 F300 ;retract the filament a bit before lifting the nozzle, to release some of the pressure\\nG1 Z+3 F' + str(speed_travel * 60) + '  ;move Z up a bit and retract filament even more\\nG90\\nG28 X Y\\nM107    ;stop fan\\nM84   ;shut down motors'"
        },
```

I was going into the `Manage Printers` → `Machine Settings` to check the GCode but nothing happens and there is this in the log:

```
file:///home/ux/.local/share/cura/resources/qml/Preferences/MachinesPage.qml:120: TypeError: Cannot read property 'onCompleted' of null
file:///home/ux/.local/share/cura/resources/qml/Preferences/MachinesPage.qml:102: TypeError: Cannot call method 'reset' of null
```

I see no other related error in the log.

Ghostkeeper, 2017-03-08:
Yes, that looks like it should work. I just tried putting this in the definition of Folger Tech FT-5. And it works. This is the g-code that I get at the end:

    ;TIME_ELAPSED:630.069422
    G1 F1500 E288.29635
    M107
    M104 S0
    M106 S255   ;start fan full power
    M140 S0   ;heated bed heater off (if you have it)
    G91   ;relative positioning
    G1 E-1 F300 ;retract the filament a bit before lifting the nozzle, to release some of the pressure
    G1 Z+3 F7200  ;move Z up a bit and retract filament even more
    G90
    G28 X Y
    M107    ;stop fan
    M84   ;shut down motors
    M104 S0
    ;End of Gcode

ubitux, 2017-03-08:
It indeed seems to work, thanks!

I guess the `Manage Printers` → `Machine Settings` menu is broken for some other reason.

closed 2017-03-08 by Ghostkeeper

fieldOfView, 2018-12-15:
@metaxis have you tried the fix by @Ghostkeeper ?

fieldOfView, 2018-12-15:
With your solution, there would need to be as many ```_mm_min``` variations as there are speed settings, which isn't great IMHO.