Ultimaker/Cura #1509: how create my plugin for cura ?

d1ms, 2017-03-08:
Hello, i have files .py from my plugin, how i can getting files .pyc ?

I read https://github.com/Ultimaker/cura-build , but i don't understand.

What's the just way compile file ?

LipuFei, 2017-03-08:
Hi @d1ms , when you import a python module, compiling `.py` to `.pyc` will be done automatically, so there is no need to do this manually.

It should just work if you put your plugin into the `plugins` directory. Make sure your `__init__.py` contains the correct information.

d1ms, 2017-03-08:
Good evening @LipuFei, i take standart plugin `LocalFileOutputDevice ` and edit only `LocalFileOutputDevicePlugin.py`. Plugin will be automatically refresh folder `__pycache__` ?

LipuFei, 2017-03-09:
Hi @d1ms, you don't need to worry about `__pycache__`. It just makes compiling python code a bit faster.

In your case, suppose you want to duplicate the `LocalFileOutputDevice` plugin and create a new one called `LocalFileOutputDeviceTest`, you can use the following steps:

1. Duplicate the `LocalFileOutputDevice` directory in the `plugins` directory
2. Edit `__init__.py` at least to rename plugin name (I named it `Local File Output Device (Test)`)
3. Save and start Cura
4. Go to menu `Preferences -> Configure Cura -> Plugins`.
5. You should be able to see the new plugin `Local File Output Device (Test)` in the list

[image]
[image]

Ghostkeeper, 2017-03-09:
If you change the .py file, Python will automatically update the version in `__pycache__` the first time you run or import the file.

d1ms, 2017-03-09:
All thanks!

closed 2017-03-09 by d1ms