Ultimaker/Cura #1510: Can't import/export printer files in Cura 2.4 (and 2.3.1)

jtzemp, 2017-03-09:
I wanted pull in a profile for a Lulzbot Taz 4 and I couldn't see any obvious way to do it. I googled around and found these links that all either talk about or show screenshots of import/export functionality:
* https://ultimaker.com/en/resources/20441-manage-profiles
* https://ultimaker.com/en/resources/21351-machines-and-profile-settings-cura-2-1-0
* https://ultimaker.com/en/community/23453-cura-230-release-bug-profiles-are-not-importing

When I pull up the manage printers page, I'm missing the import/export buttons:
[image]

I'm running MacOS 10.12.3. I had the problem first in Cura 2.3.1 and upgraded to 2.4 hoping it was just an issue with that version. The issue is exhibited in both versions.

Also, I tried quitting the app and deleting the ~/Library/Application Support/cura folder and then starting the app up again to try to rebuild preference/settings like a new install and the problem was there after nuking app prefs as well.

I'm happy to attach the cura.log, but I looked at it and nothing stood out to me as an issue.

Thanks!

nallath, 2017-03-09:
Profiles are quality profiles. These might be tied to a machine, but simply define what "normal" means with regards to settings. 

If you want a machine that's not already in Cura, you will need to make a *.def.json file that describes the properties of that machine.

fieldOfView, 2017-03-09:
> If you want a machine that's not already in Cura, you will need to make a *.def.json file that describes the properties of that machine.

... or use the "Custom FDM printer" and edit the values in Cura to match your printer.

Note that Lulzbot have (alpha-quality) builds out which include support for their printers:
http://devel.lulzbot.com/software/Cura2/

BagelOrb, 2017-07-21:
The misunderstanding has been solved.

closed 2017-07-21 by BagelOrb