Ultimaker/Cura #1511: Change request: BB core as default support extruder?

Dim3nsioneer, 2017-03-09:
The recommended settings for the UM3 in Cura 2.4.0 use printcore 1 for support material as default. However, if a BB core is selected or even detected by the Cura-UM3 link it would be more logical to have that BB core - independent whether it is printcore 1 or 2 - as default setting if support is activated. What do you think?

Ghostkeeper, 2017-03-09:
You mean, switch the print core to "BB 0.4mm" when the user activates support?

What would happen if the user is connected to a printer? Then the core would get out of sync when the user activates support. And when the print core changes on the hardware, should it deactivate support?

Dim3nsioneer, 2017-03-09:
No, no... Switch the 'support structure extruder' setting (not the core selection!) to the extruder which is actually the bb core if there is a bb core present (either selected in Cura or automatically set by the network link) and support is activated (logic behind it: if the user activates support and has the bb core mounted he wants to print with pva support). If there are only aa cores present, leave it as it is. I hope I could eliminate the remains of any clarity... ;-)

nallath, 2017-03-09:
I understand now, but i fear it's going to be quite hard to do this in the current architecture :(

Dim3nsioneer, 2017-03-09:
Take it as a challenge... ;-)

Ghostkeeper, 2017-03-09:
Perhaps a resolve function can do the job? Maybe in combination with a new setting to say that the extruder is suitable for clumping filament like PVA. But it'd be really hacky.

UltiArjan, 2017-06-26:
I think it's a good idea, I would even go so far as to also set the BB core as the default for the Brim, if a BB core is active.

Ghostkeeper, 2017-07-03:
Thinking about this again, I don't think that a resolve function is powerful enough for this... Perhaps to switch the support extruder to a support material, but not to switch the print core around based on settings.

Dim3nsioneer, 2017-07-03:
That was the original idea - I see there might have been some misunderstanding. If a BB core or alternatively PVA material is present on extruder 2, the support extruder should be extruder 2 by default.

Dim3nsioneer, 2018-05-05:
Any progress here? 3.3.1 with the S5 still selects the first extruder by default for the support material. A product which is supposed to optimize a professional workflow should select the PVA or Breakaway material as support material if present.

ChrisTerBeke, 2018-05-05:
No worries, it's somewhere on our giant list of todo's which both originate from internal product requirements and people b*tching on here...

(Not trying to offend anyone, but this is just how software development goes)

ChrisTerBeke, 2018-05-05:
Also, the default for support seems to be the 2nd extruder (although it doesn't check if there's PVA/BAM in that or not).

Ghostkeeper, 2018-06-06:
No progress yet, but our project manager put it in our top-40 where it is currently on position 28. No high priority but we'd like to add it.

Ellecross, 2019-06-06:
hey guys, it's been a while. :D To those in UM who access to this, the ticket for reference is **CURA-4015**. This new feature has been recently refined (estimated) and will probably be refined again just in case.

Ghostkeeper, 2021-05-19:
However our newest PO has rejected this as not getting priority. To give a more realistic image of when this would be fixed, we'll mark it as deferred. It's not currently in our priority list, but it's still an issue so we'll keep it open here.