Ultimaker/Cura #1516: Pause at hight (Cura 2.4-i3 FW?)

JPN3D, 2017-03-10:
I'd assume that the purpose of this is to change filament. With MK8 extruder you need to Retract 35mm change the filament and Extrude 40mm. The new filament is pushed out with current setting
M0 ;Do the actual pause
G1 E35.000000 F6000
G1 E40.000000 F200; this is too fast, a big possibility to jam and/or move the head
G1 E-35.000000 F6000; because of this

The section should look like this:
M0 ;Do the actual pause
G1 E35.000000 F6000
G1 E40.000000 F100; not F200
G1 E-3.000000 F6000; Not E35.00000
G1 Z11.000000 F300
G1 X91.551000 Y96.870000 F9000
G1 E3.000000 F6000; Not E35.00000
G1 F9000
I'd assume that the purpose of this is to change filament. With MK8 extruder you need to Retract 35mm change the filament and Extrude 40mm. Everything goes fine before the printing starts. The new filament is pushed out
M0 ;Do the actual pause
G1 E35.000000 F6000
G1 E40.000000 F200; this is too fast, a big possibility to jam and/or move the head
G1 E-35.000000 F6000; because of this

The section should look like this:
M0 ;Do the actual pause
G1 E35.000000 F6000
G1 E40.000000 F100; not F200
G1 E-3.000000 F6000; Not E35.00000
G1 Z11.000000 F300
G1 X91.551000 Y96.870000 F9000
G1 E3.000000 F6000; Noet E35.00000
G1 F9000

Ghostkeeper, 2017-03-10:
The flow values (`F200` and such) are currently hard-coded in the post-processing script. To change them you could change the script in `<Your Cura installation>/plugins/PostProcessingPlugin/scripts/PauseAtHeight.py`. You'll see the `F200` and such being inserted there.

To actually fix this, we could make the flow values into a setting as well. Should be fairly simple.

JPN3D, 2017-03-10:
Tnx, I total novice with the coding. I looked the code this line
prepend_gcode += "G1 Z%f F300\n" % (current_z + 1); is this the reason that if  redo '0' layer there will be a gap->default value to set'1'.

I managed to set new default values and commenting out those two ifs out to get this working :-)

Ghostkeeper, 2017-03-13:
I've made some adjustments for you. It's not been merged yet but it eventually will, most likely.
Take a look at this file: https://github.com/nallath/PostProcessingPlugin/raw/a1cef72c491851d2d0229530c2949534f4754ad4/scripts/PauseAtHeight.py

You can copy that file on top of your PauseAtHeight.py file. You'll get two new settings in the Pause At Height script that adjust the speed of retraction and extrusion: The F6000 and F200 that you're seeing. Keep in mind that the values you enter are multiplied by 60 to get to the g-code, since the values in the settings are in `mm/s` while the values in g-code must be `mm/min`.

JPN3D, 2017-03-14:
Hi, sorry about the delay.
The new scrip breaks something. When opening the 'Add a script' only Pause at hight BQ will appear. With the original PauseAtHigth.py there are four ones are to select.

Ghostkeeper, 2017-03-14:
Sorry, I made a mistake there. Try again: https://raw.githubusercontent.com/nallath/PostProcessingPlugin/feature_adjustable_retract_speeds/scripts/PauseAtHeight.py

JPN3D, 2017-03-14:
Yes, now this works as expected. Tnx!

(The development proposal: Extrude could be the button driven, so that filament loading could be secured. Seems that manual operation is not possible during this script)

closed 2017-03-14 by JPN3D

Ghostkeeper, 2017-03-15:
"Jogging" the printer is definitely desired but has not the priority to put it on our development agenda for now.

fieldOfView, 2017-03-15:
Lulzbot are working on it though.