Ultimaker/Cura #1517: Unable to specify a .5mm nozzle

3DKernow, 2017-03-10:
I want to print something on a UM2+ using a .5mm steel nozzle, but Cura 2.4 only gives me four options, and this is not one of them. It doesn't seem that I can write the option in.

Ghostkeeper, 2017-03-10:
Indeed, there is no interface element for custom nozzle definitions. Since the quality profiles are dependent on which nozzle is selected, using a different nozzle would prevent any profiles from showing up. There's a few things you can do:

The simplest: Adjust the line width rather than the nozzle. The actual nozzle size is only used for a few very minor things. The line width influences most of the print. So if you change the line width to 4.375 (7/8ths of the nozzle size) then your print should look as if it's been printed with a 0.5mm nozzle.

More advanced: Manually edit the nozzle size in your quality changes profile. To do this, make a new profile by modifying some setting and saving the profile. Then go to the application menu to Help -> Show Configuration Folder. Close Cura. Open the profile with a text editor in `resources/quality/<your profile name>`. Then add the setting `machine_nozzle_size = 0.5` under the values header. When you start Cura and load your profile, it should have a nozzle size of 0.5mm then under the hood (and all your settings should be tuned to that).

Expert: Actually create a definition for 0.5mm nozzles. To do that, go into your Cura installation folder and copy the 0.4mm extruder definition under `.../resources/variants/ultimaker2_plus_0.4.inst.cfg`. Modify the file name and all references to 0.4 in the file to make them 0.5. You won't have any quality profiles then, but the 0.5mm nozzle should appear in the drop-down. To enable quality profiles, you would also have to copy the 0.4mm quality profiles and change them to 0.5.

3DKernow, 2017-03-11:
Many thanks. I’ll see how this goes. Louis Turner

jackha, 2018-07-26:
Closed because of inactivity.

MarcusWolschon, 2018-07-26:
Improvements that are just not worked on get closed due to inactivity while they are still valid and important improvements?
(I have that same issue with my 1.0mm nozzle for ages now.)

jackha, 2018-07-26:
It's not on our priority list and probably never will be. @Ghostkeeper describes a workaround that seems to fix it for the original reporter, hopefully it works for you too @MarcusWolschon? 
btw The nozzle size mostly influences line widths, you can try and see if changing that will improve the print as well.

Ghostkeeper, 2018-09-06:
We normally defer these rather than closing them. This communicates that we know this is still an open problem but also communicates that we don't have time to fix them.

closed 2018-11-26 by Appesteijn

MarcusWolschon, 2018-11-26:
So nozzle sizes other then the few defaults will now never be supported? :(

Appesteijn, 2018-11-26:
@MarcusWolschon Problem is we currently don't have time to fix this and won't be doing this in the near future. Keeping it open only gives the signal that we are still looking into this (at least that is how I 'read' your reaction) and gives you a false sense of 'hope'. When closing it at least you know you should not depend on us to fix this issue.

MarcusWolschon, 2019-01-11:
Closing it communicates that you will never, ever work on it.
This is a backlog.
Only items tagged "WIP" or similar are seen as being actively worked on.

Appesteijn, 2019-01-11:
Only items tagged "WIP" or similar are seen as being actively worked on.

It is good that this is clear to you, but it might not be for everyone who visits this page. 
If this was a backlog it should have also been ordered by priority, this is not the case.