Ultimaker/Cura #1518: bed temps wrong when multi materials are (not)used

maukcc, 2017-03-10:
If you have a printer with multiple nozzles and different materials (PLA and PETG) in them, the program will choose the material with the highest bed temp to use for actual bed temp.

This is absolutely correct

but when you use only 1 extruder then it will also use the bed temp of the highest material selected for the other nozzles.
this will give unwanted results

So when only 1 extruder is used ( and it is irrelevant which extruder) then the bed temps of the other extruders should not be considered. 
I am guessing this will be a hard one to solve (even worse when looking at multi material prints which use only 1 material at the 1st layer). 
So a quick solution would be a popup when saving that reads:

You are using only 1 extruder (at 1st layer) but have multiple materials selected, is your bed temp correct?

Ghostkeeper, 2017-03-10:
Based on your questions at https://github.com/Ultimaker/fdm_materials/pull/11 I made an internal issue to have this addressed. We're planning some changes to the way that `resolve` works for the 2.6 release, so perhaps this can be picked up then too.

For reference of our developers: CURA-3499.

Ghostkeeper, 2017-03-10:
We have a function that can find all used extruders in the front-end now. I'm thinking that we modify the resolve function to simply resolve only those.

ianpaschal, 2018-02-21:
This was marked as done on Jira, and is very old, so I'm going to go ahead and close it.

maukcc, 2018-02-21:
This bug is still present in 3.2.99

ianpaschal, 2018-02-21:
@Appesteijn: 8 month old issue, marked as "Done" in Jira. Is this a `wontfix` or does it need to be reopened and evaluated?

Appesteijn, 2018-02-21:
We will (partially) fix this next sprint when you can disable an extruder and the settings will be taken from the correct extruder.

Ghostkeeper, 2018-03-13:
Having followed @jackha's work on the enabling/disabling of extruders, I think it would be completely fixed. An important reason for implementing that feature was that the resolve function would take the platform adhesion required for unused extruders into account, which is a very similar problem to this. 

I've tested this on the current master and this issue is now fixed.

closed 2018-03-13 by Ghostkeeper