Ultimaker/Cura #1522: Each printer definition file slows down the application launch

probonopd, 2017-03-11:
Currently each printer definition file slows down the application launch, even unused ones. Deleting all but the needed printer definitions greatly speeds up the launch.

Could this be changed, so that printers that have not been selected (clicked on) do not slow down the launch?

nallath, 2017-03-13:
We've made a number of changes that should limit the effect of this (lazy loading). We do need to load some data from every machine because of the add machine wizard (and others).

Ghostkeeper, 2017-03-13:
This lazy loading hasn't been put through to the material profiles though, which are a significant portion of our start-up process right now.

probonopd, 2017-08-10:
Loading machines still takes a looong time. I suspect it could be significantly shortened.

alekseisasin, 2018-05-16:
@probonopd , I think in the latest Cura version the loading process much faster than before. Now Cura loads only required printers. Adding and removing printers is also faster.

probonopd, 2018-05-16:
I can confirm this. Thanks to the Cura team for making this happen. __Big__ difference!

closed 2018-05-16 by probonopd