Ultimaker/Cura #1523: Question about variables in machine Json files

dagnall53, 2017-03-11:
Sort of not an issue, but this will make developing new machine definitions easier for amateurs like me...
I am sure that this is very simple for experienced Json writers, 

In my Machine json, I have a VERY long start up : something like this 
 "machine_start_gcode": {          "default_value": "G21 ;metric values \nG99 E0 ........lots of stuff, }",
In my Extruder Jsons, I define 
 "machine_extruder_start_code": {         "default_value": "G91 E0 ; set rel......
and "machine_extruder_end_code": {      "default_value": "G92 E0 \nG1 E-10 F600 \...

The question is, --- Can I somehow reference or use these "machine_extruder" codes in my Machine set up, to save cut and paste and typos...

I was wondering if something like: 
 "machine_start_gcode": { 
         "default_value": "G21 ;metric values \nG90 \n machine_extruder_start_code \nG92 E0 .. other stuff .......
might work to add in the relevant code ? 
but what is the "proper" way to code this please... do I need to qualify the codes with _0 and _1 at the end? Do I need quotes , ? 

Huge thanks.

awhiemstra, 2017-03-13:
The following *might* work:

```
"machine_start_gcode": {
    "value": "'G21\nG90\n{extruder_start}\nG92'.format(machine_extruder_start_code)"
}
```

Note that this was not tested and I don't even know if we properly handle value functions for the start/end gcode.

dagnall53, 2017-03-13:
Thanks for suggestion, but Cura just copies the text into the G code, 
(unless there is somethong subtle I hve missed with ", ' etc...

` "machine_start_gcode": {
         "default_value": "'G21 ;metric values\nG90 ;absolute positioning\nM82 ;set extruder to absolute mode\nM107 ;start with the fan off\nG28 ;Home all axes (max endstops) \nG29 ; auto level \nG1 X-50 Y-90 Z30 F2000; place to spit out filament\nM117 Setup extruder 1 \nT1 ;EXTRUDER 1 \n{extruder_start}\nG92 E0'.format(machine_extruder_start_code) \nM117 Printing..."
      },`

Gives this in the G code
`M109 T1 S235
'G21 ;metric values
G90 ;absolute positioning
M82 ;set extruder to absolute mode
M107 ;start with the fan off
G28 ;Home all axes (max endstops) 
G29 ; auto level 
G1 X-50 Y-90 Z30 F2000; place to spit out filament
M117 Setup extruder 1 
T1 ;EXTRUDER 1 
{extruder_start}
G92 E0'.format(machine_extruder_start_code) 
M117 Printing...
;LAYER_COUNT:85
;LAYER:0
M107`

dagnall53, 2017-03-13:
sorry, accidentally closed the issue.. Its still open for help ... please...

fieldOfView, 2017-03-13:
"value" and "default_value" are not interchangeable.

"value" expects a string that evaluates to a value, "default_value" expects a value. Only "value" gets interpreted, "default_value" does not.

Ghostkeeper, 2017-03-14:
You need to change `default_value` to `value`, or it won't get interpreted but blindly overwritten.

Once you do that, every piece of text needs to be surrounded with `'` quotes (in addition to the `"` double quotes required by JSON). You had some additional text behind the `format()` function which was not surrounded by quotes. You can perform operations with those pieces of text such as additions (`"a" + "b"` becomes `"ab"`). But you can't just write them after each other. So you'd get something like `'G21 ; metric values\\n ... {extruder_start}\\nG21 E0'.format(extruder_start=machine_extruder_start_code) + '\\nM117 Printing...'"`

You need to use double backslashes for those newlines. Both JSON and Python escape them, so you need `\\n` instead of `\n`.

Furthermore, Python's `format` function works differently. You can either use `{0}`, `{1}` in the text and don't specify named parameters in the function call, so that the first parameter gets put in place of the `{0}` and the second parameter in place of `{1}` and so on:

    '... \\n{0}\\nG92 E0'.format(machine_extruder_start_code)`

...or you can use a name between the brackets and specify a named parameter in the function call:

    '... \\n{extruder_start}\\nG92 E0'.format(extruder_start=machine_extruder_start_code)`

The second method is longer but generally considered as better code style.

Your final JSON entry should then become something like this:

    "value": "'G21 ;metric values\\nG90 ;absolute positioning\\nM82 ;set extruder to absolute mode\\nM107 ;start with the fan off\\nG28 ;Home all axes (max endstops)\\nG29 ; auto level\\nG1 X-50 Y-90 Z30 F2000; place to spit out filament\\nM117 Setup extruder 1\\nT1 ;EXTRUDER 1\\n{extruder_start}\\nG92 E0'.format(extruder_start=machine_extruder_start_code) + 'M117 Printing...'"

sedwards2009, 2017-03-14:
This talk of making the contents of the JSON files even more dependent on the peculiarities of Python, makes me a bit nervous. I would prefer an official Cura function for doing this , so that we at least have an idea about what the "language" of the expressions is, instead of saying that it is whatever Python supports and you can get away with.

My root concern is that in the future we may want to use these JSON files in a context outside Cura and possibly via a different language entirely. I'm saying let's just keep our options open here.

dagnall53, 2017-03-14:
@Ghostkeeper  (and all)
Thanks for all this advice. It is very much appreciated. Your explanation is very helpful,  (but I still have not got it working (see end of note..) )

My main reason for wanting to insert the code is to avoid repetition across the various definitions that all need to use the same code. For example, in a two extruder setup, both extruders will broadly need the same start and end codes, and the machine definition needs to re-use these as part of a start up. 
Being able to define the sequence once, and then reference it in the other definitions seemed to me a sensible way to avoid typos and errors. Any alternative easier solution would be great...

Back to the code: 
I have added the following code lines after the original... " machine_start_gcode": {  "default_value": etc etc}, stuff.: 
`"machine_start_gcode": {
	            "value": "'G21 ;metric values\\n{extruder_start}\\nG92 E0 \\n'.format(extruder_start=machine_extruder_start_code)"
				},`
Obviously this does not have everything in it yet, its just a quick test).. CURA accepts the code, and lets me use the machine, BUT it gets stuck at "preparing to slice" and does not produce any Gcode. 

The "machine_extruder_start_code" is defined in the extruder files, so is it available for the machine definition code ? 
In which order are things defined? 
or do I perhaps need to reference the specific extruder, and if so how? 

Would extruder 0 be "machine_extruder_start_code_0" if referenced by number, for example, or does it reference by "name", in which case it would be "machine_extruder_start_code_Right" ...
I have tried both and neither seem to help... I still get stuck at "Preparing to slice"... 
I have not tried  machine_extruder_start_code(0), or other variants.. 
I seem to be getting there slowly, and I think that this approach has general value to all multi extruder machines.. 

Sorry there are more questions than answers so far..

Ghostkeeper, 2017-03-14:
Ah... Hmm. Yes, this is obviously extruder-specific and can't be known in the global start g-code setting.

Well, no. You can't do `_0` at the end to get the first extruder. We do define a function `extruderValue(extruder_nr, setting_id)` though, so you could do `extruderValue(0, 'machine_extruder_start_code')`.

Repeating some of the g-code might be a smaller evil than the hacks we're doing in the JSON file now. Just something to consider.

dagnall53, 2017-03-14:
I'm getting close to just copying the Extruder codes as the lesser evil, but my start code is abouy 750 characters long, so lots of room for errors, especially whist I am refining the extruder start and ends...!.
So.. a last push.. 
` "machine_start_gcode": {
	            "value": "'G21 ;TEST OF value acceptance\\n{extruder0_start}\\n'.format(extruder0_start=extruderValue(0,'machine_extruder_start_code')"
				},`
CURA accepts this, but still gives me the Preparing to slice halt.. 
Log file says:
 `2017-03-14 12:28:52,020 - DEBUG - UM.Backend.Backend._backendLog [89]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2017-03-14 12:28:52,488 - DEBUG - CuraEngineBackend.CuraEngineBackend.slice [152]: Starting slice job...
2017-03-14 12:28:52,559 - ERROR - UM.Logger.logException [65]: Exception: Job <CuraEngineBackend.StartSliceJob.StartSliceJob object at 0x000001DDD19E4A90> caused an exception
2017-03-14 12:28:52,570 - ERROR - UM.Logger.logException [69]: Traceback (most recent call last):
2017-03-14 12:28:52,578 - ERROR - UM.Logger.logException [69]:   File "D:\2.4\build\inst\lib\python3.5\site-packages\UM\JobQueue.py", line 122, in run
2017-03-14 12:28:52,585 - ERROR - UM.Logger.logException [69]:   File "C:\Program Files\Cura 2.4\plugins\CuraEngineBackend\StartSliceJob.py", line 149, in run
2017-03-14 12:28:52,594 - ERROR - UM.Logger.logException [69]:     self._buildGlobalSettingsMessage(stack)
2017-03-14 12:28:52,603 - ERROR - UM.Logger.logException [69]:   File "C:\Program Files\Cura 2.4\plugins\CuraEngineBackend\StartSliceJob.py", line 251, in _buildGlobalSettingsMessage
2017-03-14 12:28:52,611 - ERROR - UM.Logger.logException [69]:     settings["material_bed_temp_prepend"] = "{material_bed_temperature}" not in start_gcode #Pre-compute material material_bed_temp_prepend and material_print_temp_prepend
2017-03-14 12:28:52,621 - ERROR - UM.Logger.logException [69]: TypeError: argument of type 'NoneType' is not iterable
2017-03-14 12:29:00,727 - DEBUG - AutoSave.AutoSave._onTimeout [48]: Autosaving preferences, instances and profiles
2017-03-14 12:29:57,451 - DEBUG - UM.Qt.QtApplication.windowClosed [198]: Shutting down cura
2017-03-14 12:29:58,667 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [197]: Attempting to kill the engine process
2017-03-14 12:29:58,670 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [203]: Killing engine process
2017-03-14 12:29:58,682 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [206]: Engine process is killed. Received return code 1`

dagnall53, 2017-03-14:
GOT the machine definition working...
(just need to understand why the extruderValue does not work in the extrude definitions..)  see stuff at the end of this note) 

For others wishing to emulate this... 
The crucial thing was to just use the extruderValue(nr id).
I did not need the .format stuff at all.

A working code example is: 
` "machine_start_gcode": {
	            "value": "'T0 ; select E0\\n'+extruderValue(0,'machine_extruder_start_code')+'\\nG1 E10 F100; extrude 10mm\\n'+extruderValue(0,'machine_extruder_end_code')+'\\nT1 : do the next extruder etc..\\n' "
				},`
You just need to be very careful of the \\n, and the ' that surround the text. 

This will make the machine definition start the extruders correctly, and not need modifying every time the extruder start or end codes are adjusted....

Now, just one more issue:: I have tried to do a similar thing with Extruder 1, to copy the G code from Extruder 0

` "machine_extruder_end_code": {
         "default_value": "G21 ; EX1 default end stuff\n"
		},
	 "machine_extruder_start_code": {
         "value": "extruderValue(0,'machine_extruder_start_code')"
		} `
But it does not slice... does extruderValue work in the extruder definition code or have I missed something crucial out again...?

dagnall53, 2017-03-14:
This last "extruders" code issue might be due to something else I have just noticed.. I have been getting 
`[WARNING] Couldn't load extruder.def.json file for extruder 1. Index out of bounds.`  in all my previous (working) runs. 

Any idea which "index" has been out of bounds? 

My code is a copy of the other extruders, so starts: 
` "id": "delta_y_extruder_0",
   "version": 2,
   "name": "Right",
   "inherits": "fdmextruder",
   "metadata": {
      "machine": "delta_K_2ex",
      "position": "0"
   },
   "overrides": {
      "extruder_nr": {
         "default_value": 0,
         "maximum_value": "1"
      },
`
I assume index is the extruder_nr ? What "index" do you think is out of bounds?

Ghostkeeper, 2017-03-15:
> I have been getting `[WARNING] Couldn't load extruder.def.json file for extruder 1. Index out of bounds.`

Happens all the time. I don't know what exactly causes the engine to say this but it seems to not be a problem.

> if I call "extruder X" from a machine, why does the Extruder definition ALSO need to cross reference back..

Normally we only have a reference from leaf to root, i.e. from extruder to machine. Our materials reference the machines, the quality profiles reference the materials, etc. But in this case we also need to have some ordering of the extruders. I think that's why they were referenced back as well. For quality profiles we gave them a weight by which they are sorted; I think that would've been a better solution.

dagnall53, 2017-03-17:
@Ghostkeeper  Many thanks for your help with this. 
I now have the machine definition file working and building its Start Gcode with copies of the extruder start and end codes. This avoids most of the copy and paste pitfalls.
As for the extruder files, I think a simple copy and paste of the start and end from one into the other is simple enough, so I will not explore using extrudervalue in an extruder definition any more!. (I do think there is some sort of unnecessary recursion happening, but as the copy and paste is easy, enough is enough!).

If you wish to add my two extruder, single head  Kossel definitions into the "approved set" of Cura definitions, they are at https://github.com/dagnall53/JSONS-for-Cura2_3 I am not sure what sort of testing or approval they need before being included in the distributions, but they provide an interesting example of the machine and extruder coding.

I will close this thread, with many thanks to everyone who helped.
Dagnall

closed 2017-03-17 by dagnall53

Ghostkeeper, 2017-03-21:
We typically accept pull requests for new machine definitions as long as the definitions are working correctly in Cura. We can't actually test them on the hardware (since we don't have the hardware), so we have to rely on the community to do that for us.