Ultimaker/Cura #1525: Feature request: Improved Draft Shield

Patola, 2017-03-12:
Currently the draft shield of Cura just a kind of skirt as high as the part, which is sub-optimal because for optimal hot air trapping its volume should be minimal, i.e. it should also follow the general vertical outline of the parts. Some slicers like Simplify3D and IceSL use it that way, and IceSL algorithm is very well explained in a paper: https://hal.inria.fr/hal-01141706/.

BagelOrb, 2017-03-13:
The algorithm is easy. The question is whether we find this issue important enough to fix before all the other issues. We'll have to estimate what priority this has.

Patola, 2017-03-13:
You are completely right. I for one do not think Ultimaker people should waste time on this. But feature requests like this one are a good entry point for people to become acquainted with the code of Cura and contribute.

Ghostkeeper, 2017-08-08:
Our project manager just rejected this feature because it was more than 3 months old.

closed 2018-11-26 by Appesteijn