Ultimaker/Cura #1526: Cura 2.4.0 adds M140 S60 to gcode for the printer with no heatbed.

kr15uk, 2017-03-12:
Cura 2.4.0 adds M140 S60 to gcode for the printer with no heatbed. (OSX version of Cura)
Obviously print fails within 60sec.

Temp workaround - enable heatbed for your printer, set temp's to zero, disable heatbed.

Ghostkeeper, 2017-03-14:
I can't reproduce this with the Ultimaker Original. What printer are you using? Perhaps the code is in the start g-code there.

kr15uk, 2017-03-14:
That’s a daVinci Jr 1.0 retrofitted with RAMPS and running Marlin RC7 on - so custom machine effectively.
Start/End is pretty basic.

Start:
G28 ;Home
G1 Z0.1 F6000.000
G92 E0
G1 F200 E6
G92 E0
G1 Y31.000 F6000

End:
M104 S0
Calibration cubeG92 E1
G1 E-1 F300
G28 X0 Y0
G1 Y155
M84

Heat bed is disabled (unticked) in “Manage Printers”

I’ve attached Screenshot of temp options in Cura and code it produces (line 322 is of interest)

Also probably worth to mention - I’m on macOS 10.12.3, had Cura 2.3.1 installed with this machine in there and then upgraded to 2.4.
I’ll try to remove this machine and re-add to see in it makes any difference.
Also will try to install Cura 2.4 on different machine to see if upgrade is to blame.

kr15uk, 2017-03-14:
Link to sample gcode and screenshot.

[https://drive.google.com/drive/folders/0B-4wTtCQEkGeVjFzdGI1U1FIblU?usp=sharing](url)

Ghostkeeper, 2017-03-14:
Ah, I see. It's adding the M140 at layer 1 there.
I actually fixed that 18 days ago: https://github.com/Ultimaker/CuraEngine/commit/c412a6ccfed9526fa09cf57f2dac6c44172a88ea

It's going to be repaired in Cura 2.5. See https://github.com/Ultimaker/Cura/issues/1468

closed 2017-03-14 by Ghostkeeper

kolaha99, 2019-05-20:
same problem. CURA 4.0.0 adds wrong commands for Makerbot before start-g-code

Ghostkeeper, 2019-05-20:
What printer are you using, and which g-code commands do you consider wrong?

kolaha99, 2019-07-07:
I use Makerbot Replicator. In Cura choose it from the printer list in Add Printer menu, change G-Code flavor "Marlin" to "Makerbot", save sliced G-Code file to SD Card, let gpx.exe convert it into .x3g file and gpx say "(line 10) Semantic warning: M190 cannot select non-existant heated build platform T0"
If I change option in Cura "Max Z speed" then the warning will be about unsupported M203 code.

Ghostkeeper, 2019-07-08:
> (line 10) Semantic warning: M190 cannot select non-existant heated build platform T0

Hmm, as far as I know though we never output `M190` with a `T` parameter. It doesn't make sense to select a tool chain to heat the build plate.

If you don't have a heated build plate, did you try unchecking the "Heated Bed" check mark in the machine settings?

> If I change option in Cura "Max Z speed" then the warning will be about unsupported M203 code.

Yeah, that is really specific to Marlin. We've removed the override entirely in 4.2.

kolaha99, 2019-07-08:
> If you don't have a heated build plate, did you try unchecking the "Heated Bed" check mark in the machine settings?

The Printer has the heated build plate. I have not tryed to uncheck the checkbox. 
Here is what will be added before Start-G-Code:

>;FLAVOR:Makerbot
>;TIME:14840
>;MINX:-60.861
>;MINY:-34.786
>;MINZ:0.2
>;MAXX:28.819
>;MAXY:34.795
>;MAXZ:61
>;Generated with Cura_SteamEngine 4.1.0
>M190 S90
>M104 S190
>M104 S190
>M116
>M82 ;absolute extrusion mode

Ghostkeeper, 2019-07-09:
Seems perfectly fine g-code to me. Why would it be complaining about build platform `T0`? :thinking:

kolaha99, 2019-07-09:
doubled M104 is perfectly fine?

Ghostkeeper, 2019-07-10:
No, it's not perfect, but it's at least fine. Normally CuraEngine would output:

```
M104 S190
M109 S190
```

But since Makerbot doesn't support the `M109` it turns that into a `M104`.

Shouldn't do any harm either though. I imagine it either ignores the second one or waits for the temperature to stabilize a bit longer. I thought we were talking about the `M190`? That was what your warning message was about in any case.