Ultimaker/Cura #1527: Z offset when importing STL models

Maraster, 2017-03-12:
STL Models exported from Autodesk Fusion 360 when placed on build plate  by Cura will often be offset in Z below zero. The effect is that some of the model does not print - the first several layers. The only solution I have found is to import the models into SLIC3R first, then export from that program a new STL. The resulting SLIC3R STL works properly when imported. I have noticed this happens with models not scaled correctly when first opened (i.e. Fusion's STL export seems to always use a mm scaling even if creating in inches, thus the model comes out small until scaled up). It could be related - some value related to Z placement isn't getting updated when scaling up.
[image]

fieldOfView, 2017-03-12:
Could you post or point to an stl somewhere so we can try to reproduce this?

Note: The STL file format does not specify a unit, just numbers. So Cura cannot tell if you used mm or inches.

Maraster, 2017-03-13:
Yes of course. I'm aware of the STL file format being unit-less (thanks for pointing that out of course). It is an interesting quirk of Fusion presently that if you start modeling in mm from scratch as the units, and import into Cura, the part will scale correctly without adjustment, but if you start a model in inches and then change units in Fusion (so you can measure and in the Fusion model the lengths scale correctly), then export an STL it will always need to be scaled in Cura. Just a side note - this is an Fusion file export characteristic, not Cura related. 

 I can't post this particular STL in entirety as it is for a client and proprietary, but I've noticed this happening on a few files. Let me find another one that exhibits this issue or post the lower portion of this file, if it still creates the issue - short files in Z don't always generate this problem either.

Maraster, 2017-03-13:
OK here is one - if you place this on end, a small portion of the cylinder is below zero. It seems to occur with objects that have an overhang particularly. The amount of zero offset is definitely not constant across models - it is probably some calculation error, and so model dependent, but I guess we will see. I've also added a snapshot. I'll see if I can find a model that has an even more pronounced offset. I had one model for example that was at least 2mm offset when loaded, and I could not move it up to the proper z origin. 

[25mm-OmniWheel-EdgeSpinner-01.STL.zip](https://github.com/Ultimaker/Cura/files/837046/25mm-OmniWheel-EdgeSpinner-01.STL.zip)
[image]

Ghostkeeper, 2017-03-14:
The screenshot it a bit deceptive since this is such a tiny model that it's zoomed in a lot. What you're seeing on the screenshot is the shadow that's actually hovering 0.1mm above the build plate to prevent Z-fighting.

In fact, it seems that this object is actually hovering above the build plate rather than under the build plate.
[image]

And that it's hovering above the build plate seems like a bug to me.

Ghostkeeper, 2017-03-14:
The mesh looks entirely correct in CloudCompare. Nothing weird about it, no extra vertices or something being outside of the mesh.

I'm thinking it might be some inaccuracy in the rotation matrix or something?

Maraster, 2017-03-14:
I just moved over to the new Cura 2 a few weeks ago, so I missed that option - I guess I’ll do that, though I’ve found that the slic3r import/scale/place/export of the STL works also quite reliably with automatic dropping. I will have to find a model that actually prints wrong that I can send (i.e. is not proprietary) - I have definitely found with at least three models I’m working with that it consistently prints with a negative z offset significant enough to remove entire features and walls that are more than 0.050” (i.e. ~a few mm). Interestingly enough it varies how much the offset is as you change a model. I have been making refinements to the model, and at various times when I directly import to check, the offset seems to change - however it is somehow dependent on the model features. As I tweak them it shifts the offset. I haven’t seen a 1:1 relationship that is clear from simply observing so it’s in the math most likely, it could be a random calculation issue too.

Sorry, I didn’t print this one to test, I’ll look for a better problem one to send. Also I’ll look for the shadow vs. build plate offset when looking for models that get offset - that’s a good point.

The rotation matrix or some matrix math issue - maybe a roundoff error, or something that is losing precision? It seems particularly pronounced if the part is imported to Cura and autoscales to 10000%. So if it were a calculation coming out with a scale issue, it is amplified by the size. In fact, maybe that’s the location of the problem - something related to calculating the zero z translation is happening before a scaling, or the scaling isn’t applied properly to the translation. I don’t know it would take a look at that area of the code to see. I tried importing a large version of this model and that didn’t cause the error.

Best,
Alex

Ghostkeeper, 2017-03-14:
> It seems particularly pronounced if the part is imported to Cura and autoscales to 10000%.

That is a clear hint that it's probably indeed got to do with the precision of the transformation matrices.

Maraster, 2017-03-14:
I haven’t looked over that part of Cura’s code, but it sounds like a good starting point.

Maraster, 2017-03-17:
Upon glancing at the code quickly, it looks like there is one point where we have a float defined, where the matrix operation would probably be more precise with doubles. There are also some double variables and some floats. Check out

floatpoint.h

If it’s a truncation due to variable size, switching to double float points could help. I haven’t traced through the operations that happen when a model is moved into place on the build plate yet.

fieldOfView, 2017-03-17:
@Maraster: what code are you referring to? Cura does not have a floatpoint.h

Ghostkeeper, 2017-03-17:
Aligning the models to the bottom of the bed is done before anything is sent to the engine. `floatpoint.h` has nothing to do with it, because that's in the engine.

The relevant pieces of code are here:
* https://github.com/Ultimaker/Cura/blob/master/cura/PlatformPhysics.py#L179
* https://github.com/Ultimaker/Uranium/blob/master/UM/Scene/SceneNode.py#L635

Maraster, 2017-03-17:
I see. I haven’t fully gone through the code to understand all the pieces. I’ll check out those files, thanks. I was looking over the engine, as there is one spot in the main file that performs a transformation on the meshes, but clearly that is probably not the issue. It sounds like it’s all moved before then.

Thanks for pointing to the code to look over.

Maraster, 2017-04-21:
Sorry I've been slammed lately with work for clients. I have an example of a file that offsets up in the air slightly. If I take it into slic3r, scale it and place it in the orientation to print, then export an STL, the object imports and is placed correctly, but if I open up the raw STL and place/scale in cure, it offsets. Here are the two files

[Archive.zip](https://github.com/Ultimaker/Cura/files/946213/Archive.zip)

This may be a slightly different issue but definitely related.

Ghostkeeper, 2017-04-24:
We have good reproduce steps now. When I load the two files in that zip in Cura, scale the small one up to 2500% (to match the size of the big one) and rotate it 90 degrees, then the model floats in the air like 2mm above the build plate.

Maraster, 2017-04-24:
Good to hear. 

And to bypass the problem, if you load in slic3r and do the scaling and orienting there, it definitely sits in the correct location when you open in Cura. This has become my default operation as it sometimes isn’t obvious until after a print that there has been an offset.  I guess the next thing would be to do the orienting in slic3r and then see if scaling shows the issue still. It probably would because it has to do with the z movement necessary to put the object into place, but should be checked. Then the scaling only in slic3r then orienting as another check.

Ghostkeeper, 2017-12-15:
This is fixed for Cura 3.2.

closed 2017-12-15 by Ghostkeeper

manekinekodesu, 2019-03-29:
I have the same issue in the latest stable Cura v4.0. Model exported from Fusion360 has negative zero offset about `0.2mm and the first layer is lost.
If I disable Automatically drop models... option and raise it a few millimeters then lost layer appears.
Layers also somehow drop to the plate out of raised mesh contour.
Here is the model:
[Body2.zip](https://github.com/Ultimaker/Cura/files/3023315/Body2.zip)

Ghostkeeper, 2019-04-01:
Doesn't reproduce for me. The actual mesh (a tiny hook) has a width of 0.792 according to a point-to-point measurement I did in MeshLab. When I load it in Cura it appears flat on the build plate and then slices. This produced 6 layers: 5 of 0.1mm and the initial layer of 0.3mm. Together that is 0.8mm, which is the expected behaviour.