Ultimaker/Cura #1528: Cura2.4  ignores infill percentage

Sparkyman1, 2017-03-13:
After trying to set infill percentage with "per model settings" .

changing infill percentage no longer changes the amount of infill, it just stays solid, even at 0%
also, my quality settings are all changed to default, some of my printer settings are gone, G-code start settings and nozzle size was changed to .2 but i use .5

after reinstalling cura 2.4, now it displays 0% infill in the layer view, even if infill density is set to 99%, but at 100% it makes the object solid.

nallath, 2017-03-13:
Could you provide us with a project file where this happens? I've just tried to reproduce this, but it seems to work just fine for me.

Ghostkeeper, 2017-03-14:
I'm a bit confused what the exact reproduction steps are.

You set infill percentage in the per-object settings to 0% and in the Custom settings menu to 100%? And doing this reset your settings and deleted printers?

Sparkyman1, 2017-03-14:
I'm not exactly sure what happened, but I'm guessing it has something to do with me having more than one version of cura 2.4 on my computer and fusion 360 opened up a second version of the program when I exported an STL to it. I uninstalled 2.4 and started using cura (15?) all is well with the world. I'm new to mac and not really sure if just deleting the application out of the applications folder is the right way to uninstall things so don't mind me I'll figure it out when i have time.

uwe999, 2017-07-17:
I believe I'm having the same issue with cura 2.6
To duplicate:
- load a model that easily shows the infill, select 60% "Infill density" in the main settings.
- prepare model and inspect infill visually (infill should show correctly)
- select the model and then "per model settings"
- select "infill density" and enter 30
- prepare model and inspect infill visually (infill should show correctly)
- delete the "infill density" from the "per model settings"
- prepare model and inspect infill visually (infill still shows the incorrect infill - looks like 10% to me)
- change "infill density" in main settings and they will have no effect after preparing the model
- restarting cura doesn't seem to fix it either

BagelOrb, 2017-07-21:
I cannot reproduce this in Cura 2.6 on Windows

Ghostkeeper, 2017-07-24:
Doesn't reproduce here either. The infill looks correct after deleting the setting from the per-model settings.

ChrisTerBeke, 2018-01-22:
Closing until it re-appears.

closed 2018-01-22 by ChrisTerBeke