Ultimaker/Cura #1529: Cura.exe has stopped on startup - 2.4.0 on Win10 64-bit

TangoDeltaDelta, 2017-03-13:
When trying to start Cura 2.4.0 on a very recent new install of Win10 64 bit, I see the welcome screen up to "loading interface...." then I get a "Cura.exe has stopped...." error message.  

This is on a laptop, and I've tried it both docked and undocked (to make sure it wasn't an issue with dual screens), but no luck in getting it to start.  I was running Cura (successfully) on this laptop previously under Linux Mint, but decided to rebuild the laptop with Win10, and I haven't installed very many other applications on it yet, but not being able to use Cura is kind of a big deal.

Anybody have any ideas?

DxDiag and cura.log attached.

[DxDiag.txt](https://github.com/Ultimaker/Cura/files/837210/DxDiag.txt)

[cura.log.txt](https://github.com/Ultimaker/Cura/files/837211/cura.log.txt)

nallath, 2017-03-13:
I don't see anything in the logs unfortunately. Did you also update your graphic card drivers?

TangoDeltaDelta, 2017-03-13:
I don't think there are any graphics updates to apply . . . .  This laptop has been around a few years and nothing has changed hardware-wise from what was a working Cura on Linux Mint.  
I wish Win10 would give better technical info on the error instead of what I perceive to be a dumbed-down message in an effort to be "friendly".

nallath, 2017-03-13:
Yeah same here. The point is that the drivers of your graphical card might be different, causing a crash. Those drivers are not installed on your graphical card. Changing your OS will change the installed drivers.

awhiemstra, 2017-03-13:
From your DxDiag:

>  Card name: Mobile Intel(R) 965 Express Chipset Family (Microsoft Corporation - WDDM 1.1)

This suggests you are using the default Windows graphics driver provided by Microsoft. This driver does not provide OpenGL 2 support, which is required for Cura 2. We do not handle this situation very cleanly at the moment - and efforts to improve it recently got reverted due to other Intel drivers that do weird stuff with OpenGL. Unfortunately, if Intel does not provide a proper driver, there is very little you can do.

Actpohomoc, 2017-03-23:
Same situation when working in RDP on Windows  2016 Server. Interesting that Cura 15.04.6 perfectly working there. There is very slow graphic update when move the object but works and printing w/o problem.

ianpaschal, 2018-05-23:
This is:

- more than one year old
- never reproduced
- reported for a now-unsupported version of Cura

...so unfortunately it's safe to call it dead. 😞

closed 2018-05-23 by ianpaschal