Ultimaker/Cura #1533: Cura 2.4.0 crashes on startup when a .STL model is moved or deleted.

KeithMeurer, 2017-03-13:
Cura 2.4.0 on Win 10 crashes when a file that was open in the workspace when Cura was last closed no longer exists.

Took me a while to determine what the issue was, as it was seemingly random.

Looks like it tries to start slicing when there's no object loaded.

[cura.log.txt](https://github.com/Ultimaker/Cura/files/839299/cura.log.txt)
[stderr.log.txt](https://github.com/Ultimaker/Cura/files/839297/stderr.log.txt)
[cura.cfg.txt](https://github.com/Ultimaker/Cura/files/839298/cura.cfg.txt)

nallath, 2017-03-14:
I only see errors regarding the upgrade script. 

Could you provide some clear reproduction steps for us?

KeithMeurer, 2017-03-14:
Oops, sorry about that. That would have been helpful!

1. Open Cura.

2. Open an .STL file. 

3. Close Cura without removing the object from the workspace.

4. Rename the .STL file that was open.

5. Open Cura again.  Cura will crash.

If you start Cura again, it will successfully open.

It's not a show stopper by any means, but it does slow me down as I never remember to clear the workspace before I close Cura.

nallath, 2017-03-14:
Are you very sure this has something to do with Cura? Cura does not reload things on the workspace. It doesn't even remember this, unless you specificly save them as a project file. It only loads them again if you open a project file. In that case it doesn't load it automatically and all data is inside the project file (it doesn't reference outside models).

KeithMeurer, 2017-03-14:
No, I'm not sure at all. But I don't have any special plugins loaded, and this only started happening when I upgraded to 2.4.0.

If I make sure to delete anything in the workspace before I close Cura, it doesn't crash the next time I open it.

If no one else is seeing it, I can accept the fact that it's just me. Seemed easily reproducible on my end, so I thought I'd file an issue.

Other than this, everything is working great, and Cura is still my go-to slicer.

nallath, 2017-03-14:
I really can't explain this. Could you try to make a backup of your local files, clear those and see if it still happens?

KeithMeurer, 2017-03-14:
You mean my AppData\Local\cura files? 

Okay. 

Renamed AppData\Local\cura folder to backup_cura.

Started Cura.
Picked Prusa i3 Mk2 as printer, didn't customize anything for it. (I have a Maker Select).

Loaded D:\3d\3DBenchy.stl.

Closed Cura with Benchy open.

Renamed 3DBenchy.stl to x3DBenchy.stl.

Opened Cura.  No crash.

It's definitely something with my profile. 

I can try to narrow it down, but I'd say this is probably closeable unless you want to pursue it further.

KeithMeurer, 2017-03-14:
Okay, so I renamed the new cura folder to cura_new, renamed the backup_cura folder back to cura, then cleaned up any "old" named folders, and I copied the cura\cache\definitions\2.4.0 files from the cura_new folder, as they were not the same contents in each file.

Now I can no longer reproduce the crashing, and I haven't lost any settings from my old environment.

I would say close it for sure now, as I can't duplicate the issue now either.

Thanks for your help!

Ghostkeeper, 2017-03-15:
Still strange that it happened only when renaming a file that was once opened. But let's regard it as a fluke then.

closed 2017-03-15 by Ghostkeeper