Ultimaker/Cura #1537: Cura 2.4.0 Crashes when I press open file

MarkAtZahners, 2017-03-14:
Just downloaded 2.4.0 and everytime I press Open File cura crashes. I have restarted my computer. I have uninstalled and reinstalled again. I updated my graphics drivers last week and they were working with 2.3.1

nallath, 2017-03-14:
Could you add logs?

MarkAtZahners, 2017-03-14:
[cura logs.zip](https://github.com/Ultimaker/Cura/files/842320/cura.logs.zip)

steveb3906, 2017-03-14:
same problem on a windows 10 machine running it as a user or as an administrator.

This is a new install, first time install of cura, rebooted after installation.

steveb3906, 2017-03-15:
just another comment, it only stops working when you click "File". All other menus work fine.

Ghostkeeper, 2017-03-15:
Whoa, you can actually look all the way back to the bugs we had in 15.06 in that log :grin:

I find it strange that the log says "Starting slice job..." but doesn't actually get to slicing.

himitsou, 2017-03-16:
same problem here ;-(

steveb3906, 2017-03-17:
Testing Cura 2.3.1 and it runs fine on my machine.

Hera23, 2017-03-19:
I'm having the same issue with 2.4... Everything works fine until I try to load a file and then instant crash every time.

fieldOfView, 2017-03-20:
Does ctrl+o work?
How about dropping files on the window?

himitsou, 2017-03-20:
dropping file work but for saving in a file it doesn't...

Ghostkeeper, 2017-03-21:
All right, well we aren't reproducing it on this end. I have a few more questions in addition to fieldOfView's:
* When you go into preferences, does it crash? It could be modal windows in general that make it crash.
* If not, when you go into Manage Materials and then click "Export", does it crash? This produces a save dialogue (rather than a load dialogue). Perhaps it's the file dialogues in general that make it crash.

There is probably very little we can do about it even if we reproduce the bug. The file dialogue is made by Qt, a large library we're using for the GUI. All we can really do is make a reproducible minimal test case and submit that as a bug to Qt. Unless we find a workaround in our code.

fieldOfView, 2017-03-21:
From @steveb3906's report it sounds like it is not the modal dialog, but the File *menu*. Since it only seems to happen with the File menu, could it be the MRU list?

himitsou, 2017-03-21:
préférences no crash.
material export crash

steveb3906, 2017-03-23:
Here's my testing on my Dell XPS Win10 64-Bit

ctrl-O causes crash
dropping file directly into the cura window opens the object
file --> save opens the file window but when you click "save" cura crashes
preferences window is OK
materials export window - when opens - cura crashes.

JHarding86, 2017-04-28:
I am also having this same issue on Cura 2.4.0 and 2.5.0

Anytime a save or open dialog opens, cura will crash, whether or not I click 'save' or 'open'.

nallath, 2017-05-01:
Could you post some logs? I'm rather stumped as to what could cause this.

Ghostkeeper, 2017-05-05:
Qt provides a number of dialogues, all of them slightly different. E.g. for multiple files opening, for directories, for simple files. And the bug could be in restoring the most-recently-used folder. These could result in crashes for Cura while they don't crash for JHarding.

JHarding86, 2017-06-09:
So, I came here to find out where a log file would be located so I could cappture this bug, and when I tried to save to file for the first time in a while. What ya know! It saved to file. Another interesting thing, my machine did a windows update that cleared all my recently used folders, my desktop wallpaper, all of my searches, etc etc. I think Ghoskeeper might be going down the right path.

So I think AppData/local/cura/cura.log would be the one you are looking for?

nallath, 2017-06-09:
Yep!

Ghostkeeper, 2017-06-10:
Note that for Cura 2.6 the correct file is located in `AppData/roaming/cura/cura.log`.

awhiemstra, 2017-06-12:
`AppData/roaming/cura/2.6/cura.log` even.

JHarding86, 2017-06-16:
Ok, I think I got the log. This is the most immediate ERROR message:

2017-06-15 18:52:18,753 - ERROR - SliceInfoPlugin.SliceInfo.run [58]: An HTTP error occurred while trying to send slice information: HTTP Error 500: Internal Server Error

Appears as some sort of logging to an external server is taking place? Well, trying and failing?

[cura.txt](https://github.com/Ultimaker/Cura/files/1079435/cura.txt)

nallath, 2017-06-16:
That error is one that is caught & handled. It just logged so we know it happend. I'm pretty sure it has nothing to do with your issue.

Appesteijn, 2017-07-14:
Maybe: https://bugreports.qt.io/browse/QTBUG-41416 ?

luckshot, 2017-07-21:
My coworker and I did some experimentation (company bought a new 3D printer and we're tasked with being its keeper).  His computer has this trouble and mine doesn't.  Hopefully these observations might help:

I observed everything work for about an hour.  When we came back from lunch this dialog bug began to happen.  Whatever the bug is it doesn't necessarily happen immediately.  Maybe a corrupt internal file?

The difference between our computers is largely software:  I'm a software engineer and he's a mechanical engineer, so I have an up-to-date VS2015 and QT developer tools already present, so I did not elect to install the vs2015 redist that comes with the Cura installer.  His computer needed it to run.

It appears not to be the File menu so much as the file dialog that causes the problem.  The crash happens in both open and save menus.

As a workaround, double-clicking the file to open via file association avoids the open file dialogue and the file loads without error.  If there was a way to save without the dialogue I suppose we'd have a fully valid workaround.

Drag-and-drop seems to have no trouble either.

Sorry for the verbose post, I'm hoping something here might help.

luckshot, 2017-07-21:
I've attached the cura.log file from the aforementioned use case.  User clicked the save button on the lower left, the save dialog came up, displayed properly, then the cursor goes to "thinking" state followed by the Windows crash handler.

[cura.txt](https://github.com/Ultimaker/Cura/files/1166902/cura.txt)

Ghostkeeper, 2017-08-02:
> I observed everything work on his PC for about an hour. When we came back from lunch this dialog bug began to happen. Whatever the bug is it doesn't necessarily happen immediately.

That's actually very interesting. That might indicate that there could be some cache file that you can delete or Cura can rewrite to fix the issue.

The log shows nothing really interesting as far as I could find. Perhaps there was only this snippet:

2017-07-21 15:18:26,987 - CRITICAL - cura.CrashHandler.show [39]: An uncaught exception has occurred!
    2017-07-21 15:18:27,040 - CRITICAL - cura.CrashHandler.show [42]: Traceback (most recent call last):
    2017-07-21 15:18:27,068 - CRITICAL - cura.CrashHandler.show [42]:   File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 685, in event
    2017-07-21 15:18:27,085 - CRITICAL - cura.CrashHandler.show [42]:   File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Qt\QtApplication.py", line 261, in event
    2017-07-21 15:18:27,103 - CRITICAL - cura.CrashHandler.show [42]:   File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Event.py", line 189, in call
    2017-07-21 15:18:27,123 - CRITICAL - cura.CrashHandler.show [42]:   File "D:/2.6/build/inst/lib/python3.5/site-packages/UM\Signal.py", line 315, in __performEmit
    2017-07-21 15:18:27,143 - CRITICAL - cura.CrashHandler.show [42]:   File "D:/2.6/build/inst/lib/python3.5/site-packages/cura\CuraApplication.py", line 1325, in _readMeshFinished
    2017-07-21 15:18:27,159 - CRITICAL - cura.CrashHandler.show [42]: AttributeError: 'NoneType' object has no attribute 'width'

It seems to refer to this line of code: https://github.com/Ultimaker/Cura/blob/153d1f2d0c6b0be84b168150136bad97a5547ecb/cura/CuraApplication.py#L1353

It's not the error itself though. We can't really log that since the Python environment crashes before we get the chance to.

If you're a software engineer and can spare the time, perhaps you can run his Cura through GDB and get us a stack trace?

Ghostkeeper, 2017-08-04:
Another user eventually found out that it started working again when he disabled his Adobe and Google Drive extensions in Windows. These don't seem to be playing nicely with uncommon file dialogues.

TheM0thership, 2017-08-11:
Ghostkeeper, can you be more specific about what that user disabled? Did he disable all file extensions associated with Adobe products?

Ghostkeeper, 2017-08-18:
@TheM0thership As far as I've heard, he disabled everything related to Adobe and Google products, yes. It's probably caused by one of the two, but we don't know which.

Ghostkeeper, 2017-10-16:
Our project manager removed this from our planning since it's more than 12 weeks old.

JHarding86, 2017-10-16:
That’s unfortunate as I’m still having this exact problem. Not to mention the new version of cura can’t print a single one of my old stl files. Rather disappointing.

himitsou, 2017-10-16:
So do I 
Same problem with latest version. 2.7

Thomas Péjoan & Sabine Pouget

jackha, 2018-07-26:
If anybody has this problem with the latest version, 3.4.1, then I'd be happy to reopen this.

MaxxDelusional, 2018-08-10:
I have this problem in 3.4.1

Ghostkeeper, 2018-08-13:
The [QTBug](https://bugreports.qt.io/browse/QTBUG-33119) that seems to cause this is not yet solved either.

gabry1609, 2018-09-01:
@MaxxDelusional me too, having the same issue.

lancef1, 2019-03-20:
I am having this issue in Cura 4.0.  I have had this issue in every version of Cura i have ever installed starting with 3.2.  Any time the windows explorer dialogue box opens Cura crashes.  If I try and open a file from the menu, it crashes.  If I try to import a settings profile, it crashes.  The only way to open a file properly is to drag and drop from windows explorer into an already open Cura window.

rtw819, 2019-08-11:
NOTE: If you have a Dell Laptop/PC, this may be useful information or something to at least look at trying.  We had this same crash issue when opening/saving files on my wife's Dell laptop with Cura ver. 4.1.0 and even the latest 4.2.1 -- see Cura Git issue #4220 for more specific details.

The issue was fixed after disabling 3 Dell Backup & Restore Shell Extension services (used Nirsoft's shexview.exe). Here are 2 of 3 disabled:

DBROverlayIconBackuped.DBROverlayIconBackuped
and
DBROverlayIconNotBackuped.DBROverlayIconNotBackuped

The third (disabling may not have been needed) but it was named something very similarly and likely related, so I went ahead and disabled it as well.

Thanks to all whom have posted in these git issues.  Hope this helps.

Ghostkeeper, 2020-03-05:
Last week, the QTBUG that this is caused by was closed as "out of scope": https://bugreports.qt.io/browse/QTBUG-33119?focusedCommentId=499937&page=com.atlassian.jira.plugin.system.issuetabpanels%3Acomment-tabpanel#comment-499937

They say it's an issue that they can't fix. Seems that they are blaming Windows on it.

closed 2023-10-14 by github-actions[bot]