Ultimaker/Cura #1540: Cura 2.4 "heating failed" in 3rd layer of raft

bond4u, 2017-03-14:
3 prints in a row Marlin 1.1.0-rc8 told "heating failed print halted".
Thought it's intial layer temp problem, but setting it to 0 didn't make a difference.
Switched back to Cura 2.3.1 and it's printing fine.
[files.zip](https://github.com/Ultimaker/Cura/files/842226/files.zip)

kr15uk, 2017-03-14:
Had the same issue. Apparently will be fixed in v2.5.

Let me guess you don't have heated bed?!
If so then go to printer settings, enable heated bed, in temp settings set bed to 0 and you should be all good (you can disable bed back in settings if you like).

bond4u, 2017-03-17:
Thanks! That did the trick.
Tried Marlin RCBugFix branch, but it didn't make a difference.
Set all four temp settings to same value and that didn't work either.

BagelOrb, 2017-07-21:
fixed

closed 2017-07-21 by BagelOrb