Ultimaker/Cura #1541: Plugins are loaded in random order

AmedeeBulle, 2017-03-14:
PluginRegistery uses os.listdir() to get plugins.  
Order is not defined with this call, so depending on how the files are in the directory block, you get different order for loading plugins

Small test to illustrate:
```
>>> import os
>>> os.listdir("/opt/cura24/usr/bin/plugins/plugins")
['XRayView', 'UltimakerMachineActions', 'GCodeProfileReader', 'CuraProfileWriter', 'ConsoleLogger', 'LocalFileOutputDevice', 'FileHandlers', 'Views', 'GCodeWriter', 'MachineSettingsAction', 'LayerView', 'X3DReader', 'SliceInfoPlugin', 'VersionUpgrade', 'PostProcessingPlugin', 'Tools', 'USBPrinting', 'PerObjectSettingsTool', 'LegacyProfileReader', 'CuraProfileReader', 'SolidView', 'UM3NetworkPrinting', 'XmlMaterialProfile', 'ChangeLogPlugin', '3MFReader', 'ImageReader', 'CuraEngineBackend', 'AutoSave', 'Doodle3D', 'RemovableDriveOutputDevice', 'UpdateChecker', '3MFWriter', 'FileLogger']
>>> os.listdir("/mnt/usr/bin/plugins/plugins")
['3MFReader', '3MFWriter', 'AutoSave', 'ChangeLogPlugin', 'ConsoleLogger', 'CuraEngineBackend', 'CuraProfileReader', 'CuraProfileWriter', 'Doodle3D', 'FileHandlers', 'FileLogger', 'GCodeProfileReader', 'GCodeWriter', 'ImageReader', 'LayerView', 'LegacyProfileReader', 'LocalFileOutputDevice', 'MachineSettingsAction', 'PerObjectSettingsTool', 'PostProcessingPlugin', 'RemovableDriveOutputDevice', 'SliceInfoPlugin', 'SolidView', 'Tools', 'UltimakerMachineActions', 'UM3NetworkPrinting', 'UpdateChecker', 'USBPrinting', 'VersionUpgrade', 'Views', 'X3DReader', 'XmlMaterialProfile', 'XRayView']
```

This would be perfectly OK if there were no dependency, but as PostProcessingPlugin depends on CuraEngineBackend, so if it is loaded first it will crash:

```
2017-03-14 19:50:14,674 - ERROR - UM.Logger.logException [65]: Exception: Error loading plugin PostProcessingPlugin:
2017-03-14 19:50:14,687 - ERROR - UM.Logger.logException [69]: Traceback (most recent call last):
2017-03-14 19:50:14,693 - ERROR - UM.Logger.logException [69]:   File "/build/2.4/build/inst/lib/python3/dist-packages/UM/PluginRegistry.py", line 87, in loadPlugin
2017-03-14 19:50:14,713 - ERROR - UM.Logger.logException [69]:   File "/opt/cura24/usr/bin/plugins/plugins/PostProcessingPlugin/__init__.py", line 21, in register
2017-03-14 19:50:14,749 - ERROR - UM.Logger.logException [69]:     return {"extension": PostProcessingPlugin.PostProcessingPlugin()}
2017-03-14 19:50:14,780 - ERROR - UM.Logger.logException [69]:   File "/opt/cura24/usr/bin/plugins/plugins/PostProcessingPlugin/PostProcessingPlugin.py", line 43, in __init__
2017-03-14 19:50:14,784 - ERROR - UM.Logger.logException [69]:     self.scriptListChanged.connect(Application.getInstance().getBackend().forceSlice)
2017-03-14 19:50:14,785 - ERROR - UM.Logger.logException [69]: AttributeError: 'NoneType' object has no attribute 'forceSlice'
```

Ghostkeeper, 2017-03-15:
We specifically make sure that ConsoleLogger and CuraEngineBackend are loaded first: https://github.com/Ultimaker/Cura/blob/138c3db26f836955da8b7b1c1dab4dcec204212b/cura/CuraApplication.py#L462

So it should not load PostProcessingPlugin first. How did you get that?

AmedeeBulle, 2017-03-15:
Oh I see!

The problem is with the Linux AppImage: sys is "frozen" there, so the pre-load of these 2 is by-passed, and they are loaded in random order.

In one case the order is purely alphabetical (and we are lucky that  ConsoleLogger and CuraEngineBackend are early in the alphabet), in the other one it is more random and CuraEngineBackend comes very late...

Ghostkeeper, 2017-03-21:
Hmm, I don't see why it would bypass the pre-loading of these two when sys is "frozen", but still load the other plug-ins even if it's frozen. Do you know this, @awhiemstra?

awhiemstra, 2017-03-21:
Good question. I do not remember. Though the console logger should be in the block, as it may not be available on a frozen build. But the engine can be pulled out of it.

AmedeeBulle, 2017-03-21:
I don't see why any of them should be in the block...  
I have re-packaged the Linux build for 2.4 with these 2 outside the block and it runs perfectly.

Maybe I am missing something, but I don't see what.

Ghostkeeper, 2017-03-22:
Worth noting that the `loadPlugins()` call right below it eventually does call `loadPlugin` without checking for `sys` being frozen.