Ultimaker/Cura #1542: Importing a material fails to add material

mattpaul, 2017-03-15:
Cura v2.3.1, Mac OS X 10.12.2

Importing a previously exporting custom material fails to add the material to the list of materials in Manage Materials interface or in the Material drop-down UI.

This is extremely frustrating as the work around requires you to manually remember and re-enter per-material parameters manually after app launch.

Ghostkeeper, 2017-03-15:
We made some changes to that for 2.4. Could you try again with the more recent version?

markwal, 2017-03-17:
It doesn't seem to work in 2.4 either.  Plus, duplicate a material, edit, quit and restart and the added material is gone.

Ghostkeeper, 2017-03-21:
I couldn't reproduce this :(

My steps were as follows:
1. Add UM3 printer.
2. Duplicate the PLA material. Give it the label BLA.
3. Export the material (to bla.xml.fdm_material).
4. Delete the material.
5. Import the material again.

Expected result:
* The BLA material gets added to the list of materials.

I also tried to reproduce it with the UM2+ and UMO, and it just seems to work fine. What printer were you using?

markwal, 2017-03-21:
UM2. I tried it with these steps:

1. Duplicate generic PLA call it Raptor PLA, change brand from Generic to Makergeeks
2. Export the material
3. Quit
4. Restart

Expect material to still be in the list, but it isn't, so then:

5. Import material

Result: It doesn't

Ghostkeeper, 2017-03-21:
The material disappearing after restart is a duplicate of #1104, #1404, #1410, #1452, #1471 and #1508. That's not the issue I was reading here though. But perhaps it is causing this issue.

Ghostkeeper, 2017-03-21:
For me those reproduce steps split off after step 4, because the material is still in the list after the restart :confused:

mattpaul, 2017-03-21:
I continue to experience the issue after updating to the latest version of Cura. I have a different printer, the Prusa i3 MK2. Upon clicking 'import' after relaunching Cura and selecting the file the dialog box says "Successfully imported material file://..." but the material cannot be found in the list of materials or popup menu.

I don't require import/export functionality for interoperability or sharing with other instances. Alternatively if Cura simply remember the custom/duplicate materials I create and edit that would work too. My printer uses 1.75mm filament but all of the default materials are 2.85mm which means I need to duplicate/edit/customize material upon every single app launch which makes for a rather poor user experience. Other than that I really like the software. Hoping my feedback can help the team to prioritize accordingly. Thanks!

markwal, 2017-03-21:
While I understand that mattpaul pointing out the issue with a non-Ultimaker printer using an non-Ultimaker filament diameter could reasonably cause this to be prioritized very low (so I hope you'll ignore mattpaul's hope for it to be prioritized accordingly and prioritize unaccordingly. :-)

But since you can't reproduce, it might be difficult.  I wonder if what happens is that there is a settings state that has the new material in the list, but not showing for some reason and the import collides with the old hidden material.  Perhaps something to do with having other settings stacks sitting around.  I'll try with a clean settings folder and see what happens.

markwal, 2017-03-22:
I looked at the workaround for issue #1104 and deleted "<Compatible>True</Compatible>" from my fdmmaterial and then it imported as "RaptorPLA #4".  I suspect the other 3 are previous failed imports that are not shown.  Also my original "label" was "Raptor PLA" so maybe that space is messing things up.

Ghostkeeper, 2017-03-22:
> I wonder if what happens is that there is a settings state that has the new material in the list, but not showing for some reason and the import collides with the old hidden material.

That's my suspicion too... Your new test seems to affirm that.

Ghostkeeper, 2017-03-22:
Still cannot reproduce (on master at least), even with Prusa i3 MK2... :confused:

markwal, 2017-03-23:
Maybe it's fixed post 2.4 then?  A few things I've noticed:

1. If you cd %LOCALAPPDATA%\cura\materials and edit each of the fdm_material's found there to delete the line: <compatible>True</compatible> from them, then they'll magically re-appear in the UI (don't even need to import).
2. cura.log has entries like these:
...\InstanceContainer.py (addMetaDataEntry [131]: Meta data with key "compatible" was already added.
3. I have created several dual extrusion printer instances (Ultimaker Original Dual and Ultimaker 3 [you tried that one]) and they give log entries: Multi-extrusion printers are not supported.

Conjecture: since the material settings are somehow per extruder, perhaps you need to have a dual extrusion printer somewhere and/or more than one printer [because they each have an extruder train] and there's some comment in XmlMaterialProfile.py

line 224 in my case:
# Compatible is a special case, as it's added as a meta data entry (instead of an instance).  Hanging off of hotend.

Ghostkeeper, 2017-03-23:
We're bringing the 2.5 beta online in exactly two hours. It's been branched off master two weeks ago, so unless something significant changed in those two weeks...

Your point 2 is interesting. Might be some conflict in our implementation.
Point 3 is a different problem. It's actually reported by the engine, not the front end. It doesn't seem to give problems at the moment.

fieldOfView, 2017-03-23:
Point 3 is reported by the machine settings action plugin. There's a work in progress PR that changes this

markwal, 2017-03-23:
I reread my comment just now and realized the line I suggested to delete had tags in it that this markdown editor strips out:

\<compatible\>True\</compatible\>

Zaniix, 2017-03-24:
I am having this issue too, I can not create a custom material, if will disappear as soon as I close the program and I can not import  removing the True statement did not help

Zaniix, 2017-03-24:
I deleted all instances of Cura from my PC and installed 2.4 from a fresh download and same issue I can not make it save any new filaments.  I guess its time to try 2.5 beta

Zaniix, 2017-03-24:
2.5 beta does not have this issue.

closed 2017-03-27 by Ghostkeeper