Ultimaker/Cura #1543: Arduino knock-off Mega2560 boards not recognised as a printer (CH340/CH341)

kr15uk, 2017-03-15:
Arduino knock-off Mega2560 boards which are using CH340/CH341 are not being recognised as a printer. Just says printer not connected.
Logs doesn't show anything when machine is being plugged in via USB.
Also tried to plug machine in 1st and then start up Cura - no change.

Drivers are all fine - Arduino IDE, Repetier host etc works fine.

Board has Marlin RC7 on.

Cura 2.4.0 on macOS 10.12.3

nallath, 2017-03-15:
USB printing is shaky at best. Mostly because the newer ultimaker machines (2 & 3) don't use USB printing at all. This means that Ultimaker won't spend time / resources on fixing it. 

If others want to fix this, they are of course free to do so. Pull requests will be reviewed and accepted as normal.

bond4u, 2017-03-30:
CH34x drivers are needed.
Arduino IDE does not recognize boards without driver, too.

girtsf, 2017-04-11:
Cura 2.5.0-beta here. Monoprice Maker Select V2. I had attempted to get it working on mac, but looks like I finally got it to find the printer on Windows.

I had to hack up USBPrinterOutputDeviceManager.py's getSerialPortList to add 'or "VCP" in values[0]:' to the comparison, as my ports would show up in registry as "\Device\VCP0" instead of "USBSER" that it was looking for. And I did mess with the USBPrinterOutputDevice.py as well (commented out stk500v2 detection).

Ghostkeeper, 2017-04-13:
If that VCP-hack works, we could consider merging that if you make a pull request. Removing stk500v2 is out of the question though.

ivanporro, 2017-05-01:
Hope this helps someone to work this out.

I have a Megatronics on Windows, detected as COM3 on \Device\VPC0.
I had to do three hacks on Cura 2.5.0 to avoid any connection error in the debug log:
1) add "or VPC" in getSerialPortList() to detect the VPC0 which was on COM3
2) force the iterator i on port list to start from 1 instead of from 0. This helps to skip COM1 and go to COM3
3) force 250.000 as the first choice for baudrate in _getBaudrateList()

With this done if I load a model, and hit print, the printer start. However, monitor still says "printer not connected".

rmcolbert, 2017-05-02:
I submitted PR #1778 to add the change to support VCP type ports in Windows.

closed 2018-08-16 by jackha