Ultimaker/Cura #1546: Cura 2.4 for Linux fails to keep connection open

jwatte, 2017-03-16:
I'm trying to set up a Linux-based dedicated 3D printing station. I am using a Printrbot Simple Metal (which uses a Printrboard rev F5) and a Zotac Zbox mini-PC running Ubuntu 16.04. I am using the AppImage for Cura 2.4. This printer previously worked fine with Cura 15.04 on Windows.

The printer shows up on /dev/ttyACM0, and the software starts to recognize it, but then it immediately gives up, because it's surprised to get 0 bytes back when the file descriptor is ready to read. This is actually totally legit behavior for serial ports that are set in polling mode! I would expect the code to just loop back around and try again if this happens (perhaps with a small usleep of 100 microseconds or whatever.)

Here is the cura.log file. I could not find a stderr.log file.

Is there something simple I can do to work around this? Because I run from within an AppImage, I can't find any convenient Python files to edit to try to patch this.

[cura.log.zip](https://github.com/Ultimaker/Cura/files/846633/cura.log.zip)

Ghostkeeper, 2017-03-16:
This is the relevant line in that log:

> 2017-03-15 20:55:36,638 - ERROR - USBPrinting.USBPrinterOutputDevice._readline [616]: Unexpected error while reading serial port. read failed: device reports readiness to read but returned no data (device disconnected or multiple access on port?)

It is Python's `serial` module that does the check and gives this error. Seemingly we're not the only ones with this problem: http://arduino.stackexchange.com/questions/16866/pyserial-and-serial-serialutil-serialexception-device-reports-readiness-to-read

However, everywhere I look I'm seeing people respond that it is or was a kernel bug, not the intended behaviour.

jwatte, 2017-03-16:
Ubuntu 16.04 uses kernel 4.4, so the previous recommendation for updating the 3-series kernel is years out of date.

The thread linked to by the thread you linked to seems to verify that the bug is "back" in 16.04:
https://github.com/pyserial/pyserial/issues/17

So, that doesn't actually solve my problem. I can't patch the Cura because it's built into an appimage.
I looked at the GitHub repository, but it seems Cura has a lot of dependencies to build.
Is there a simplified "download all dependencies and set them up for a build" script that can kickstart building my own copy?

Ghostkeeper, 2017-03-17:
Look into these repositories:
* https://github.com/Ultimaker/cura-build-environment
* https://github.com/Ultimaker/cura-build

Ghostkeeper, 2017-03-17:
If you know of a workaround we can try... But we can't test our workaround unless we have a way to reproduce the bug ourselves too.

ender210, 2017-03-26:
You don't have to use Cura 2.4. Go into the pb archives and get the last version of Cura 15. Then you can make the changes to the pyserial.py on your system. Thats where I am at the moment. I just don't what changes to make.