Ultimaker/Cura #1547: Slicing hangs indefinitely for some printer models

phile314-fh, 2017-03-16:
Application Version: 2.4.0
Platform: Linux NixOS 17.03
Qt: 5.7.1
PyQt: 5.8.1
Display Driver: Mesa DRI Intel(R) HD Graphics 520 (Skylake GT2), Mesa 17.0.0

Steps to Reproduce:
Start Cura, add Printer "Ultimaker 2+", Load stl model.

Actual Results:
"Preparing to slice..." text appears in lower right status field, stays there forever.

Expected results:
Slicing should finish.

Additional Information:
Bug seems to be only triggered by certain printers:
Working:
- Ultimaker Original+
- Ultimaker 2

Broken:
- Ultimaker 2+
- Ultimaker 3

Suspicious log entry:
<pre>
2017-03-16 17:21:05,608 - CRITICAL - cura.CrashHandler.show [35]: TypeError: unorderable types: NoneType() > int()
2017-03-16 17:21:05,622 - CRITICAL - cura.CrashHandler.show [32]: An uncaught exception has occurred!
2017-03-16 17:21:05,623 - CRITICAL - cura.CrashHandler.show [35]: Traceback (most recent call last):
2017-03-16 17:21:05,623 - CRITICAL - cura.CrashHandler.show [35]:   File "/nix/store/mk9gdgfgwy1q5ydympwwrpy2w9ghr41k-uranium-2.4.0/lib/python3.5/site-packages/UM/Qt/Bindings/MainWindow.py", line 178, in _render
2017-03-16 17:21:05,624 - CRITICAL - cura.CrashHandler.show [35]:     view.beginRendering()
2017-03-16 17:21:05,624 - CRITICAL - cura.CrashHandler.show [35]:   File "/nix/store/jx09rfihvaxg9myx0rsq8wzk97agp2xg-cura-2.4.0/lib/cura/plugins/SolidView/SolidView.py", line 49, in beginRendering
2017-03-16 17:21:05,624 - CRITICAL - cura.CrashHandler.show [35]:     multi_extrusion = global_container_stack.getProperty("machine_extruder_count", "value") > 1
</pre>

Full Log: https://gist.github.com/phile314-fh/33f013ffacb105c11521a7fb4eebef45
Original NixOS issue: https://github.com/NixOS/nixpkgs/issues/23951

hroncok, 2017-05-03:
Reproduced.

Application Version: 2.5.0
Platform: Linux Fedora 25
Qt: 5.7.1
PyQt: 5.7.4
Display Driver: Mesa DRI Intel(R) HD Graphics 5500 (Broadwell GT2)

Steps to Reproduce:
Start Cura, add Printer "Ultimaker 3", click "Finish". Try to select Ultimaker form the list of printers on the right.

Actual Results:
The "platform" is gone, the printer name does not show up. Loading STL file does not show it.

Expected results:
Should show the Ultimaker 3 platform, should say "Ultimaker 3" in the select box. Should render STLs.

Additional Information:
Switching back to "Ultimaker 2" works.

Keep repeating in the  log:
```
2017-05-03 17:38:21,419 - CRITICAL - cura.CrashHandler.show [36]: An uncaught exception has occurred!
2017-05-03 17:38:21,419 - CRITICAL - cura.CrashHandler.show [39]: Traceback (most recent call last):
2017-05-03 17:38:21,419 - CRITICAL - cura.CrashHandler.show [39]:   File "/usr/lib/python3.5/site-packages/UM/Qt/Bindings/MainWindow.py", line 183, in _render
2017-05-03 17:38:21,420 - CRITICAL - cura.CrashHandler.show [39]:     view.beginRendering()
2017-05-03 17:38:21,420 - CRITICAL - cura.CrashHandler.show [39]:   File "/usr/lib/cura/plugins/SolidView/SolidView.py", line 53, in beginRendering
2017-05-03 17:38:21,420 - CRITICAL - cura.CrashHandler.show [39]:     multi_extrusion = global_container_stack.getProperty("machine_extruder_count", "value") > 1
2017-05-03 17:38:21,420 - CRITICAL - cura.CrashHandler.show [39]: TypeError: unorderable types: NoneType() > int()
```

A gif explaining what's happening:

[image]

Ghostkeeper, 2017-05-05:
I can't reproduce this.

My reproduction steps are:
* Clear all settings.
* Start Cura.
* Add Ultimaker 2.
* Add Ultimaker 3.

It switches to an Ultimaker 3 as expected. The build volume renders fine. It slices. Everything looks normal.

What reproduction steps are you using? Could you try clearing your settings before you try?

hroncok, 2017-05-05:
What is the best way to clear all settings?

When I delete `.config/cura/cura.cfg` and `.config/Unknown Organization/cura.cfg`, it asks me to add a printer on startup, but I still see the old printers in the selection box.

hroncok, 2017-05-05:
Ah, `.local/share/cura/`.

hroncok, 2017-05-05:
Repeated:

 * Clear all settings.
 * Start Cura.
 * Add Ultimaker 2.
 * Add Ultimaker 3. Click "Finish" on the printer connection menu.
 * Select Ultimaker 3 from the select box on top right.
 * Symptoms as described before are repeated.

```
2017-05-05 14:54:35,003 - CRITICAL - cura.CrashHandler.show [39]: Traceback (most recent call last):
2017-05-05 14:54:35,003 - CRITICAL - cura.CrashHandler.show [39]:   File "/usr/lib/python3.5/site-packages/UM/Qt/Bindings/MainWindow.py", line 183, in _render
2017-05-05 14:54:35,003 - CRITICAL - cura.CrashHandler.show [39]:     view.beginRendering()
2017-05-05 14:54:35,004 - CRITICAL - cura.CrashHandler.show [39]:   File "/usr/lib/cura/plugins/SolidView/SolidView.py", line 53, in beginRendering
2017-05-05 14:54:35,004 - CRITICAL - cura.CrashHandler.show [39]:     multi_extrusion = global_container_stack.getProperty("machine_extruder_count", "value") > 1
2017-05-05 14:54:35,004 - CRITICAL - cura.CrashHandler.show [39]: TypeError: unorderable types: NoneType() > int()
```

hroncok, 2017-05-30:
I can also reproduce this on Fedora 26, 5.8.0, PyQt 5.8, Python 3.6.1.

Ghostkeeper, 2017-05-31:
The reproduce steps couldn't be more clear, but it's still not happening on my side. After adding the Ultimaker 3, it is already selected. I can select it again but nothing happens then because it's already selected, and I get no errors. :disappointed:

hroncok, 2017-06-28:
I cannot longer reproduce this on 2.6.1 :tada:

Ghostkeeper, 2017-07-03:
It's still happening sometimes. I just got a report of it today from the System Testing team. But we still don't know how to reproduce it and it seems to happen only 2% of the time or so.

ianpaschal, 2018-02-26:
Based on that ^ I don't believe this is significant enough to keep open.

closed 2018-02-26 by ianpaschal