Ultimaker/Cura #1550: Navigation cube

probonopd, 2017-03-18:
It would be nice if Cura would have a navigation cube known from many other 3D applications.

[image]

foreachthing, 2017-03-18:
Nice one, @probonopd!
And if you're at it, add a possibility to enable/disable perspective (orthographic) view. This would make layer-view much easier to over view.

fieldOfView, 2017-03-18:
Unfortunately Autodesk has a patent on the navigation cube.

probonopd, 2017-03-19:
I'd be surprised if the general idea could be patented (might be wrong though, ianal). Do you have a link? Maybe we'd need to make it a bit different then, but still something similar.

foreachthing, 2017-03-19:
Use arrows and planes...

probonopd, 2017-03-19:
Actually if we could have this existing UI element interact with the mouse like a navigation cube, then that'd do the job:

[image]

nallath, 2017-03-20:
@probonopd Software patents. Which would mean we can't distribute to america anymore. 

Even if a court would deem it invalid, it would still cost a whole lot of money, so I'd rather not take the risk.

BagelOrb, 2017-07-21:
Let's consider the OP's last idea to circumvent the patent.

fieldOfView, 2017-07-21:
Caveat: for some printers, this is in the center of the buildplate.

Ghostkeeper, 2017-08-02:
It'd also be a bit weird for elliptical build plates to click on the sides.

Ghostkeeper, 2017-11-25:
This has been implemented with several buttons at the top of the screen in 3.1.

closed 2017-11-25 by Ghostkeeper