Ultimaker/Cura #1552: M204 does not set accelration in Repetier

brickZA, 2017-03-19:
Using Cura 2.4.0 build from thopiekar Ubuntu PPA. If I adjust print/travel acceleration under speed settings for a Repetier flavour printer, it generates (for a setting of 100mm/s^2) `M204 S100`, which is the same as for the RepRap (marlin/sprinter) flavour. 

For repetier (such as my Wanhao Duplicator i3 v2.1 factory ROM seems to use), M204 sets PID parameters (according to https://github.com/repetier/Repetier-Firmware/blob/master/src/ArduinoDUE/Repetier/Repetier.ino)

The nearest equivalent seems to be:

  - M201 - Set max acceleration in units/s^2 for print moves (M201 X1000 Y1000)
  - M202 - Set max acceleration in units/s^2 for travel moves (M202 X1000 Y1000)

smartavionics, 2017-03-20:
Hi, this change has been done already, Cura 2.5 will use M201 & M202 for Repetier flavor gcode.

brickZA, 2017-03-20:
Great, I'll see if I can build the 2.5 branch for myself so long. I have a question though: Marlin also seems to have an identical M201/202 (set  print maximum acceleration), while Marlin's M204 is "Set Starting Acceleration". This seems to describe different behaviour, although I can't claim to understand how it differs.

smartavionics, 2017-03-20:
The 2.5 beta builds could start appearing soon so that would save you the effort of building it yourself. Don't forget to back up your configuration files before you try it out, just in case it does something weird.
Sorry, can't help with the M204 thing, I have no experience with that.

brickZA, 2017-04-02:
Can confirm that this is fixed in 2.5 Beta.

closed 2017-04-02 by brickZA