Ultimaker/Cura #1555: Cura 2.4 stuck on "Slicing..."

SuperFrodo, 2017-03-21:
2017-03-21 20:19:07,976 - DEBUG - CuraEngineBackend.CuraEngineBackend.slice [152]: Starting slice job...
2017-03-21 20:19:07,977 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [197]: Attempting to kill the engine process
2017-03-21 20:19:07,981 - DEBUG - UM.Backend.Backend._createSocket [208]: Previous socket existed. Closing that first.
2017-03-21 20:19:07,998 - DEBUG - UM.Backend.Backend._logSocketState [167]: Socket state changed to Listening
2017-03-21 20:19:08,001 - ERROR - UM.Backend.Backend.startEngine [80]: Unable to find backend executable: C:\Program Files\Cura 2.4\CuraEngine.exe
2017-03-21 20:19:11,752 - DEBUG - UM.Mesh.MeshData.calculateNormalsFromVertices [379]: Calculating normals took 0.022022724151611328 seconds
2017-03-21 20:19:11,758 - DEBUG - STLReader.STLReader.read [63]: Loaded a mesh with 677118 vertices
2017-03-21 20:19:12,071 - DEBUG - UM.Mesh.MeshData.approximateConvexHull [347]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.3122987747192383 seconds. 458 input vertices. 458 output vertices.
2017-03-21 20:19:12,184 - DEBUG - UM.FileHandler.ReadFileJob.run [59]: Loading file took 0.4889676570892334 seconds
2017-03-21 20:19:12,497 - DEBUG - UM.Mesh.MeshData.approximateConvexHull [347]: approximateConvexHull(target_count=1024) Calculating 3D convex hull took 0.3108203411102295 seconds. 458 input vertices. 458 output vertices.
2017-03-21 20:19:13,150 - DEBUG - CuraEngineBackend.CuraEngineBackend.slice [152]: Starting slice job...
2017-03-21 20:19:13,151 - DEBUG - CuraEngineBackend.CuraEngineBackend._terminate [197]: Attempting to kill the engine process
2017-03-21 20:19:13,154 - DEBUG - UM.Backend.Backend._createSocket [208]: Previous socket existed. Closing that first.
2017-03-21 20:19:13,169 - DEBUG - UM.Backend.Backend._logSocketState [167]: Socket state changed to Listening
2017-03-21 20:19:13,171 - ERROR - UM.Backend.Backend.startEngine [80]: Unable to find backend executable: C:\Program Files\Cura 2.4\CuraEngine.exe
2017-03-21 20:19:13,210 - DEBUG - CuraEngineBackend.CuraEngineBackend._onStartSliceCompleted [287]: Sending slice message took 0.058568477630615234 seconds

My System:
Windows 10 Home 64-bit
Intel i7-4790, 16GB DDR3
Nvidia GTX 980

Anything else you need to know, just ask.

I am unable to get Cura to slice. The model loads, but it just say's "Slicing..." and the layer view does,'t show anything except a transparent image of the model. I've tried different machines as well to see if it's just my settings, but no joy.
Googling my issue only really led to one or two posts, but I didn't find a solution. Someone was having this problem with Cura 2.3.1, and found they could slice on version 2.3, but I haven't been able to try 2.3 because whenever I attempt to download it from the Ultimaker website, nothing happens.

I have tried turning off my firewall, turning off anti-virus, applying "run as administrator" to both Cura and CuraEngine.exe, as well as trying different windows version compatibility settings for Cura.

fieldOfView, 2017-03-21:
Have you tried different models, or just one model?
Have you tried moving the model around on the buildplate?

awhiemstra, 2017-03-21:
> 2017-03-21 20:19:13,171 - ERROR - UM.Backend.Backend.startEngine [80]: Unable to find backend executable: C:\Program Files\Cura 2.4\CuraEngine.exe

That one is the relevant line here. What kind of anti virus software are you running? Because previously this problem was caused by anti virus software deleting the backend executable for some reason.

SuperFrodo, 2017-03-21:
> Have you tried different models, or just one model?
Have you tried moving the model around on the buildplate?

Yes and yes.

> That one is the relevant line here. What kind of anti virus software are you running? Because previously this problem was caused by anti virus software deleting the backend executable for some reason.

I use Windows Defender and the payed version of Malwarebytes. I tried running it without anti-virus running, but nothing has been deleted, CuraEngine.exe is still there. I will try reinstalling with anti-virus turned off.

SuperFrodo, 2017-03-23:
Well, I just installed Cura 2.4 on my Surface Pro 2 and it works fine. Meanwhile, even after fully uninstalling my old version of Cura and 2.4, reinstalling 2.4 without anti-virus running it still won't slice. I guess for now I can at least use Cura on the Surface, but it won't work as well for larger prints that will take a while to load/slice with this computer having an i5.

Ghostkeeper, 2017-03-23:
We run the command `C:\Program Files\Cura 2.4\CuraEngine.exe` as a system command. When this gives a `FileNotFoundError` in Python we display the message you're seeing.

You could try this for yourself by pressing the Windows key, typing `cmd` to enter the command line, and entering the command `C:\Program Files\Cura 2.4\CuraEngine.exe` there.

SuperFrodo, 2017-03-24:
> We run the command C:\Program Files\Cura 2.4\CuraEngine.exe as a system command. When this gives a FileNotFoundError in Python we display the message you're seeing.
>
>You could try this for yourself by pressing the Windows key, typing cmd to enter the command line, and entering the command C:\Program Files\Cura 2.4\CuraEngine.exe there.
>
>With the Surface you'll probably run out of memory in no time ;)

I just tried running that in cmd:
http://i.imgur.com/Ag2qryC.png

The Surface seems to be doing fine for $480AUD. Every now and then CuraEngine.exe crashes in the middle of slicing, restarting the slice usually results in success, but it's worth it when the old version of Cura I was using was creating artefacts while 2.4 is giving perfect prints. And the settingsssssssssssssss.

Ghostkeeper, 2017-03-24:
Ah, sorry. It needs to be in quotes because of the spaces in the path:

    "C:\Program Files\Cura 2.4\CuraEngine.exe"

SuperFrodo, 2017-03-24:
Here's what cmd spat out.

Microsoft Windows [Version 10.0.14393]
(c) 2016 Microsoft Corporation. All rights reserved.

C:\Users\Robert>"C:\Program Files\Cura 2.4\CuraEngine.exe"

Cura_SteamEngine version DEV
Copyright (C) 2014 David Braam

This program is free software: you can redistribute it and/or modify
it under the terms of the GNU Affero General Public License as published by
the Free Software Foundation, either version 3 of the License, or
(at your option) any later version.

This program is distributed in the hope that it will be useful,
but WITHOUT ANY WARRANTY; without even the implied warranty of
MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
GNU Affero General Public License for more details.

You should have received a copy of the GNU Affero General Public License
along with this program.  If not, see <http://www.gnu.org/licenses/>.

usage:
CuraEngine help
        Show this help message

CuraEngine connect <host>[:<port>] [-j <settings.def.json>]
  --connect <host>[:<port>]
        Connect to <host> via a command socket,
        instead of passing information via the command line
  -j<settings.def.json>
        Load settings.json file to register all settings and their defaults
  -v
        Increase the verbose level (show log messages).

CuraEngine slice [-v] [-p] [-j <settings.json>] [-s <settingkey>=<value>] [-g] [-e<extruder_nr>] [-o <output.gcode>] [-l <model.stl>] [--next]
  -v
        Increase the verbose level (show log messages).
  -p
        Log progress information.
  -j
        Load settings.def.json file to register all settings and their defaults.
  -s <setting>=<value>
        Set a setting to a value for the last supplied object,
        extruder train, or general settings.
  -l <model_file>
        Load an STL model.
  -g
        Switch setting focus to the current mesh group only.
        Used for one-at-a-time printing.
  -e<extruder_nr>
        Switch setting focus to the extruder train with the given number.
  --next
        Generate gcode for the previously supplied mesh group and append that to
        the gcode of further models for one-at-a-time printing.
  -o <output_file>
        Specify a file to which to write the generated gcode.

The settings are appended to the last supplied object:
CuraEngine slice [general settings]
        -g [current group settings]
        -e0 [extruder train 0 settings]
        -l obj_inheriting_from_last_extruder_train.stl [object settings]
        --next [next group settings]
        ... etc.

In order to load machine definitions from custom locations, you need to create the environment variable CURA_ENGINE_SEARCH_PATH, which should contain all search paths delimited by a (semi-)colon.

Ghostkeeper, 2017-03-24:
Thanks. Well, that seems to work properly. But when Python does it from a separate process the file suddenly doesn't appear to exist... :(

nallath, 2017-07-27:
Closing as this isn't a cura issue, but a overzealous virus scanner. We do attempt to register cura with them, but a few always seem to decide that Cura is a trojan / virus / bla

closed 2017-07-27 by nallath