Ultimaker/Cura #1558: Crash: AttributeError: 'NoneType' object has no attribute 'x'

ErikDeBruijn, 2017-03-21:
Name tag Anke
Version: 2.5.99-master.20170314
Platform: Darwin-15.6.0-x86_64-i386-64bit
Qt: 5.7.0
PyQt: 5.7.1

Exception:
```
Traceback (most recent call last):
  File "/Users/ultimaker/build/master/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 1036, in mergeSelected
  File "/Users/ultimaker/build/master/build/inst/lib/python3.5/site-packages/cura/CuraApplication.py", line 1036, in <listcomp>
AttributeError: 'NoneType' object has no attribute 'x'

I was removing objects (dual merged objects) and things were moving around.
```

Ghostkeeper, 2017-03-22:
Line numbers have moved a bit since then, but this seems to be the relevant line of code: https://github.com/Ultimaker/Cura/blob/72db7772f538a32201c9bcce6b0aa3b1c97cd540/cura/CuraApplication.py#L1060

So it seems like `mesh.getCenterPosition` can return None?

Ghostkeeper, 2017-03-22:
Can't reproduce this :( I tried merging, grouping, nested grouping. And then removing those models and moving them around. Perhaps we'll need some more specific reproduce steps.

Ghostkeeper, 2017-03-24:
Apparently this was fixed here: https://github.com/Ultimaker/Cura/commit/ecbf0e1e281cbbf11aee95214b5165a59d211a0c

closed 2017-07-27 by nallath