Ultimaker/Cura #1560: How to set Z-offset of second nozzle?

karabas, 2017-03-22:
Prusa like 3d printer with dual  independent extruders. 
I need to set Z-offset when I switch to second extruder.
But there are only X and Y offsets in config files.
How to set?

Ghostkeeper, 2017-03-23:
There's no feature like that yet. I'll have it added to our tracker.

fieldOfView, 2017-03-23:
How does the UM3 handle it then? Purely in the firmware?

fieldOfView, 2017-03-23:
I hope to finish support for Machine Settings for multiextrusion printers, which will let you edit extruder switch gcode snippets. I think those snippets could include the Z offset too?

karabas, 2017-03-23:
Its impossible via firmware. Even if you remove nozzle and put it back there is a possibility of its height changing...

karabas, 2017-03-27:
As a workaround I use M218 command in Marlin RC-bugfixed. After some code hacks it works. But currently does not save via M500

Ghostkeeper, 2017-06-26:
Our product manager just rejected this issue because it had been inactive for three months.

closed 2018-11-26 by Appesteijn