Ultimaker/Cura #1561: 3MF files and dual extrusion

w00kie, 2017-03-23:
Hello,

So I'm on Cura 2.4 and exporting an assembly from Solidworks to 3MF format and loading it in Cura for dual extrusion on my Ultimaker 3. Clicking on the model will select the whole assembly and after much fumbling around I found that Ctrl+click will let me select specific bodies of the model.

Now my first problem is that, coming from my experience with STL where each part of the assembly was loaded as a separate object that I could select independently, with 3MF all I can select are bodies, not whole parts. In my example model I have a logo inserted into a bigger shape. The whole logo is one single part in Solidworks composed of multiple bodies (the letters of a word for example). So if I want to have a different color for that logo insert, where with STL I would select the whole part as one and apply the material choice, with 3MF I must select each and every separate bodies.

While I could live with that first problem, the second problem that crops up is that when you select multiple bodies with Shift+Ctrl+click, the `Per Model Settings` is not available. So I must select just one body, change the material, then go to the next and repeat, potentially on many dozens of little pieces around the model.

Regards,
Francois

Ghostkeeper, 2017-03-23:
The 3MF format has a concept of mesh groups, and in Cura this is translated as something you typically want to select and interact with together.

Setting per-object settings on groups has been requested before, I think. In any case, I made a ticket in our tracker to get this planned.

Ghostkeeper, 2017-06-26:
That ticket just got rejected by our product owner for being inactive for 3 months.

nallath, 2017-07-27:
Setting extruder on groups is now possible. So it's now fixed.

closed 2017-07-27 by nallath