Ultimaker/Cura #1562: 3MF files and custom coordinate systems

w00kie, 2017-03-23:
So admittedly I don't know if this is a bug in Cura or in Solidworks' 3MF exporter.

I'm using Solidworks to design stuff and it has that annoying issue of having the Y-axis for height by default. So I usually do the trick of adding a new coordinate system with Z-axis for height and selecting this one in the export options. This works fine for STL but today I wanted to try out the 3MF format and found that whatever output coord system I select, when opening the 3MF file on Cura it keeps the default orientation of Solidworks.

Ghostkeeper, 2017-03-23:
Cura doesn't use a different coordinate system for 3MF. We do apply all transformations in the 3MF as well though, so perhaps those undo the changes made by Solidworks to transform the coordinate system, if they didn't also transform their transformations?

nallath, 2017-03-23:
It's a bug with solidworks. 

3MF has a clearly defined coordinate system; Z is upward from the buildplate.

w00kie, 2017-03-23:
Do you recommend another model viewer that supports 3MF where I could do some tests to compare how they handle model orientations?

Ghostkeeper, 2017-03-24:
Try OpenJSCAD: https://openjscad.org/

w00kie, 2017-03-30:
I actually work for Dassault Systemes (parent company of Solidworks) and got my hands on a [Solidworks-Ultimaker cookbook](https://github.com/Ultimaker/Cura/files/881647/SW-UM-cookbook.pdf) that was published for Solidworks World 2017. It advises to use `Edit > Reset All Model Transformations` when opening a 3MF file in Cura.

It seems to work for me although doing away with the extra step would be nice.

Ghostkeeper, 2017-03-30:
Our reference implementation for 3MF has been 3D Builder by the way. If Cura displays the file differently than 3D Builder, we've previously considered that a bug (bar some features that we or they don't support).

nallath, 2017-03-30:
The ultimaker cookbook gives this as an advice because solidworks does it wrong.

BagelOrb, 2017-07-21:
This is not a Cura bug, but a SolidWorks bug.

closed 2017-07-21 by BagelOrb