Ultimaker/Cura #1563: Incorrect path rom prime tower. BUG

karabas, 2017-03-23:
In dual nozzle mode and offset on first extruder. Prime tower is on.
After printing prime tower head goes firstly not to model but to the point of the tower BUT with no offset applying. The same with both extruders.

Ghostkeeper, 2017-03-24:
After printing a prime tower, it'll still be in the prime tower. You say it's then going to the corner of the prime tower, but it should already be in the corner of the prime tower because it just printed the prime tower. Or am I visualising this wrongly? Perhaps a picture or some example g-code could illustrate what's going wrong?

karabas, 2017-03-27:
It seems as  prime tower last printing coordinate looses offsets (coordinates of prime tower). So head goes far beyond the prime tower, as if prime tower located in another place, and after  to model Situation is similar to delta bug of prime tower shadow. Or maybe "extruder_prime_pos_CCCC" settings suddenly turn on.

Ghostkeeper, 2017-03-27:
I'm not reproducing this. I've sliced a dual-colour print with a printer where the first extruder had 7mm y-offset and the second extruder had 18mm x-offset. No strange travel moves. Goes directly to the model after the prime tower of the layer is completed.

karabas, 2017-03-28:
[image]
[image]

You can see that additional side pathes appears. Marked with red arrows.
The only difference is Prime Tower on/off. If I change pos of prime tower red pathes are moves correspondently.

Ghostkeeper, 2017-03-29:
I agree that this definitely looks like a bug. I've tracked this in our internal issue tracker (CURA-3612).

karabas, 2017-04-05:
My  distance between nozzles is very big - ~87mm along the X axis.  You may see on the pics that it is exactly the same distance to left and right wrong movement points.

Ghostkeeper, 2017-06-26:
Our product manager just rejected this because it was inactive for 3 months.

ianpaschal, 2018-02-27:
CURA-3612 is marked as fixed as of August 2017.

Ghostkeeper, 2018-03-19:
No, it's not. It's put in the category for deferred issues.

closed 2023-06-24 by github-actions[bot]