Ultimaker/Cura #1565: latest version from master ppa broken for ubuntu 16.04

StefanKi, 2017-03-23:
I am using the cura master ppa on ubuntu 16.04.

Cura doesn't start.
When the little loading window dissapears, nothing more happens. The main window doesn't show up.

In the log I can see following message:
file:///usr/share/cura/resources/qml/OpenFilesIncludingProjectsDialog.qml:4 module "QtQuick" version 2.6 is not installed

Ubuntu 16.04 has Qt5.5 and with that QtQuick version 2.5.
Ubuntu 16.10 has Qt5.6 and with that QtQuick version 2.6.

So Cura will only work with ubuntu 16.10 and up in the future? This would be a very very bad idea, because ubuntu 16.04 is the latest LTS version and I don't think about upgrading to 16.10 or 17.04 in the near future.

StefanKi, 2017-03-24:
I tested this on a second pc, and there is the same problem.

Before the update to the currently latest cura master everything worked fine.
After the update cura wants QtQuick 2.6 (which isn't available on ubuntu 16.04) and stops working.

I know, that it is possible to update to Qt 5.6 on ubuntu 16.04, but this workaround messes up the system.

fieldOfView, 2017-03-24:
Note that the PPA is not officially supported by Ultimaker. Ultimaker supplies AppImages instead. The PPAs are a community project.

StefanKi, 2017-03-24:
This confuses me.
Whats the difference between the downloaded version of the ppa and a version that would be supplied as AppImage? I think this would be the pretty much same build.
If the Bug isn't fixed, then a supplied AppImage in the future could have this bug and so no one with ubuntu 16.04 would be able to use the software.

I thought I would help when testing the master branch and posting bugs...

Is there a better place to share such infos or bugs?

nallath, 2017-03-24:
By all means use the master version. I think fieldOfView is just trying to warn you that you're using an unstable version (just in case that wasn't an intentional action). We've had quite a few people who thought the master PPA was stable.

StefanKi, 2017-03-24:
Yeah I know that I am using an unstable version and I won't blame anyone for any bugs there...

Ghostkeeper, 2017-03-24:
We do support Ubuntu 16.04. As far as I know, we still support 15.10 and up (because earlier versions still used Python 3.4).

The culprit seems to be this commit from @LipuFei: https://github.com/Ultimaker/Cura/commit/892225d39848b2fce3d62a5ecd6a17187da43341
This was later amended by this commit from @nallath: https://github.com/Ultimaker/Cura/commit/a8cec631b497d92696a05da24c3e5f06d0ffb1bf

fieldOfView, 2017-03-24:
I was also under the impression that the AppImage comes with all the dependencies included in the AppImage, making this issue irrelevant for the AppImage. But I could be totally wrong because I have never used the AppImage.

thopiekar, 2017-03-24:
Already heard from someone else. What you can do is using an additional PPA (which could broken your whole installation) and provide this dependencies. Since dependencies are often changing I don't care about versions in my packaging.
I personally won't fix or do something about it, thus I recommend you to use the stable PPA for now.

For me, if it is not possible to lower the needed version of QtQuick, then we can close it.
I can then disable builds for Xenial on the next/upcoming release in the stable PPA.

StefanKi, 2017-03-24:
@Ghostkeeper
Yes that's the problem.
"Import QtQuick 2.6" must be pulled back to "Import QtQuick 2.5"
The question is why this was set to "Import QtQuick 2.7" in first place - hopefully there are no things that are not working anymore with <2.6?

@thopiekar
I know about the additional ppa and I will test that, but it is not the best solution.
If QtQuick can't be set to 2.5 for whatever reasons, it will break the native compatibility to ubuntu 16.04 forever.

Ghostkeeper, 2017-03-24:
I've further lowered the requirement: 0f0b8cd5dd2a40a673b5fc444797007b804c9db5

StefanKi, 2017-03-24:
Glad to hear that!
I changed the code locally and it is starting up again without problems.

StefanKi, 2017-03-24:
Problem solved - closed...

closed 2017-03-24 by StefanKi

thopiekar, 2017-03-24:
👌

awhiemstra, 2017-03-27:
The problem here is actually unclarity about QtQuick versioning in my opinion. It is very hard to figure out what version of QtQuick matches with what version of Qt, except for whatever the latest version is.

StefanKi, 2017-03-27:
When searching around for the problem above I found this one: https://forum.qt.io/topic/50948/how-to-find-in-documenation-the-qtquick-versions-for-a-qt-version/2

If it's true, it should be easy. Until now it didn't turn wrong for me.

awhiemstra, 2017-03-27:
Except that:

http://doc.qt.io/qt-5.6/qml-qtquick-item.html

mentions `import QtQuick 2.5`

and

http://doc.qt.io/qt-5/qml-qtquick-item.html

mentions `import QtQuick 2.7`

so it seems they skipped a version somewhere...