Ultimaker/Cura #1566: unable to convert a C++3D 'QVector' instance to a Python object

Labern2, 2017-03-23:
Version: 2.5.0-BETA
Platform: Windows-8.1-6.3.9600
Qt: 5.7.1
PyQt: 5.7.1

Exception:
TypeError: unable to convert a C++3D 'QVector' instance to a Python object

nallath, 2017-03-24:
I know we had issues with the QVector, but I thought those were fixed. Guess I was wrong.

Ghostkeeper, 2017-03-24:
I don't seem to be able to reproduce this. My steps were:
* Add UM3 printer.
* Load two items on the bed (I used two cubes with different sizes).
* In the preferences, enable "Center camera when item is selected".
* Activate layer view. I don't think this has anything to do with it, but I did it anyway.
* Click repeatedly on either cube. Sometimes slow. Sometimes fast.

No errors yet. Should I do something differently?

Labern2, 2017-03-24:
I get the Oops! a fatal exception error message pop up every time I click on something with "Center camera when item is selected" is enabled in preferences. 
Have tried in solid view and with only 1 item on the bed and every time I click on a part it brings up the error message and force closes Cura.

nallath, 2017-03-24:
I know that @jackha and @awhiemstra worked on this issue. Both of them are off today, so you will have to wait until monday. Sorry about that.

awhiemstra, 2017-03-27:
This is actually a regression in PyQt 5.7.1 that we primarily observed on Windows. We tried reverting to PyQt 5.7.0 but that failed. We did not find a proper solution yet, unfortunately. We should probably make sure to keep center camera disabled for the release.

Ghostkeeper, 2017-03-29:
This only happens on Windows as far as we know.

Ghostkeeper, 2017-07-09:
Ah, by the way, this has been fixed this in Cura 2.7. It was a bug in Qt that got fixed for us when we upgraded to Qt 5.8.

closed 2017-07-09 by Ghostkeeper

b4545, 2017-07-16:
Fantastic