Ultimaker/Cura #1568: Cant change Language

ManuelW77, 2017-03-24:
In Cura 2.5 Beta i cant change the language. I did a clean install of Cura.

nallath, 2017-03-24:
Our beta's don't have correct languages. We tend to change a lot of strings in development, so they need re translating. We use the time of the beta to get the translations.

closed 2017-03-24 by nallath

ManuelW77, 2017-03-24:
Ok, make sense.

Is there a open project for translation?

Ghostkeeper, 2017-03-24:
I'd expect language changing to still mostly work. Just the texts that are new or changed since the last version would remain Developer English. I just tried and it seems that it indeed doesn't seem to switch languages at all, even though the preference is set properly.

We'll have a string freeze somewhere next week. After that I'll be sending the translation templates off to the translation bureau, some of Ultimaker's distributors, and to the guys in the community that made our Portuguese and Russian translations.