Ultimaker/Cura #1572: Feature Request: Zoom In Mouse Pointer Direction

Labern2, 2017-03-24:
Most CAD packages zoom into the area where the mouse pointer is pointing when you scroll the mouse wheel.

This allows you to point at a part you want to zoom into and scroll the wheel to get a closer look in that direction.
This would be really nice and make the user experience faster when you have multiple parts on the bed or when checking layer view.
It would also make Cura more intuitive for CAD designers.

Ghostkeeper, 2017-03-24:
Ah, currently we just zoom in towards the centre of the screen. I think this would be very useful indeed.

Ghostkeeper, 2017-03-24:
Also quite fun to implement, I think :)

Labern2, 2017-03-24:
I'm glad you will enjoy implementing it. Because I will have fun using it.
I wish I could program. It has taken me like 6 months to try and add an extra endstop and I still can't figure it out.

ChrisRiddell, 2017-03-25:
+1 for this from me as on a Macbook from what I can find you can't move the build plate around so if you zoom in you only get to see up close to where it takes you.

BagelOrb, 2017-03-26:
@ChrisRiddell SHIFT+left click drag doesn't work on Mac?!

@Ghostkeeper perhaps it's better to move forward / backward instead of zooming. It can be quite annoying to not be able to get a part larger on your screen because you are far away and reached the max zoom level.

fieldOfView, 2017-03-26:
Cura already does a dolly move instead of a true zoom

BagelOrb, 2017-03-26:
Then why is there a max zoom level?

fieldOfView, 2017-03-26:
The dolly pivots around an origin (which does not have to be on the center of the buildplate; eg if you click on an object, that origin becomes the center of the object). The max zoom level is basically the minimum distance from that origin. 

I _think_ the reason for that minimum distance is that the closer you get, the smaller the steps you make to zoom in further. If the minimum distance were 0, then there would be rounding issues when zooming back out (ie: zoomFactor * currentDistance when currentDistance = 0 makes it impossible to zoom out).

BagelOrb, 2017-03-26:
Alright, so then i guess this feature request would involve placing the
pivot at the place the mouse is pointing at, without changing the camera
position and then moving the camera toward the new pivot.

<https://github.com/Ultimaker/Cura/issues/1572#issuecomment-289295276>, or mute
the thread

ChrisRiddell, 2017-03-27:
@BagelOrb It's is now not sure why it stopped working last time I had it open :/

Ghostkeeper, 2017-03-27:
Actually, there is no zoom limit. You can just keep scrolling and you'll eventually "punch through" to the other side.

BagelOrb, 2017-03-29:
Actually zooming is quite a difficult subject. When rotating you would want
to rotate around a pivot, while when zooming the pivot is an invisible
boundary point where you cannot get past when trying to get closer.

In blender you can switch to a fly mode in which you move forward, strafe
up/down/left/right and the camera rotates around its own center by using
the mouse. I don't know whether that would work for cura, though.

Maybe we should look at how solidworks handles the camera. For user
friendly interface blender is our worst bet! :O

Ghostkeeper, 2017-06-26:
Our product manager just rejected this issue because it was inactive for 3 months.

BagelOrb, 2017-06-26:
> Well that's dissapointing as it would have really improved the user
> experience.

Thing is that there's so much ideas about what to do next that we have to
make a choice and disregard some options.

Dim3nsioneer, 2017-06-26:
@Ghostkeeper @BagelOrb 3 months of inactivity do not necessarily mean nobody is interested in the feature. A lot of people might quietly like it. I just put some wake up call onto the forum to make people understand they should support ideas here actively.
So +1 from me for Labern's idea here.

BagelOrb, 2017-06-26:
Let's hope enough people respond with +1's so that we can get a good
picture of what people generally want..

nickthetait, 2017-06-26:
Hint hint, anyone that is interested in this feature please comment here.
I am one of these people :+1:

nallath, 2017-06-26:
I'll have a look tomorrow after work to see how hard it is to build.

BagelOrb, 2017-06-27:
The way in which Blender does this is by making the amount by which you
move the camera forward depends on the object which is pointed to by the
mouse pointer.

That way you don't go through an object when trying to view a detail of a
model.

fieldOfView, 2017-06-27:
On a personal note, I would not take Blender as an example of how to do mouse/keyboard interaction.

BagelOrb, 2017-06-27:
Haha true that.

Visually it's the worst program, but otherwise it has quite good UI. It has
shortcuts for everything and the look around options are quite nice. It's
just that you can't find anything and right mouse button is ffed.

Labern2, 2017-06-27:
In terms of cad and 3d model navigation. Solidworks is pretty much the benchmark. Other programs like Onshape imitate this style of navigation as its the most common.

BagelOrb, 2017-06-27:
True.

Just checked. Solidworks zooms in while keeping what the mouse is pointing
at at the same location on the screen. Zooming out is just zooming out
based on the middle of the screen.

Labern2, 2017-06-27:
The first part of this clip shows the zoom action. So you can point to a part of the model you would like to look at closer and scroll the mouse wheel to zoom to that section. Yes, it does bring it to the center of the screen.
https://youtu.be/JyuZXZddMUQ

nallath, 2017-06-27:
See pull request. Had a bit of "dead" time to implement it.

nallath, 2017-06-27:
Right now it also zooms out in the direction of the mouse. What do you guys think? Is that behavior that should stay that way, or should we always zoom out by means of the center?

DDDirk, 2017-06-27:
That sounds good. I sometimes use this to pan if I have the mouse in my hand. Say I'm zoomed in to the left and want to pan to the right, then I move the mouse to the right and zoom out and then in. 
Btw, being a keyboard afficionado I do this by moving the mouse, but zooming in and out with + and -. Of course, I would much more like to do this purely with the keyboard. Right now (Cura 2.5.0) the arrow keys only rotate. It woud be nice if that would pan if combined with, say, Ctrl.

Labern2, 2017-06-27:
Yes that's good. I also use it to quickly pan.

nallath, 2017-06-28:
Hmm, it's not working quite right; The rotation point isn't updated quite right.

nallath, 2017-07-27:
This has been fixed. Whoo.

closed 2017-07-27 by nallath

herrdeh, 2018-02-28:
Great I found this new feature! - To spread the word, I'd like to suggest to make it the default behaviour!

BagelOrb, 2018-02-28:
+ 1

ianpaschal, 2018-02-28:
I hate such behavior but maybe @LukeChen-Ultimaker has an opinion?

Labern2, 2018-03-01:
It needs a bit of tweaking before it should be made default. It needs to slow down the zoom factor the closer to the item you get and also the view needs to be restrained somehow to you don't lose the build platform.