Ultimaker/Cura #1573: Feature Request: Horizontal Expansion on Initial layer

Labern2, 2017-03-24:
The squish factor that helps make prints stick to the glass also leaves the print with a sharp edge on the bottom.
If you could adjust the horizontal expansion and enter a - amount (eg -0.3) then this will remove the sharp edge and make the print square on the bottom. I have done tests with this by adding 2 stl's. One was the height of the Initial layer which I used per object settings to enter the -0.3 horizontal expansion. merged the 2 stl's and got a print with a square edge on the base.
This would allow a more aggressive initial layer and not distort the prints base to get strong adhesion.

Ghostkeeper, 2017-03-24:
That's basically what Brim does, isn't it?

Labern2, 2017-03-24:
No. It's kinda the opposite of brim. 
the initial layer is designed to squash onto the glass. The more squash the stronger it sticks to the glass.
the squashing action makes the line width wider and it ends up wider than the actual parts dimensions. If you print a part with skirt selected, you end up with a tiny brim from the squashed initial layer.
This can effect parts that need dimensional accuracy like an internal threaded hole and it makes the part sharp on the bottom.
But if you use horizontal expansion more as horizontal contraction on the initial layer you can remove the tiny brim or sharp edge.
Pictures shown are print placed upside down. One showing the sharp edge the other was with my modified horizontal expansion settings set to -0.3. Not perfect examples but illustrates my point.
[image]
[image]

Ghostkeeper, 2017-03-27:
Ah, I see. It's to compensate elephant feet. Well, that could help. I'm making a feature ticket and we'll see what the planning allows.

Ghostkeeper, 2017-06-26:
Apparently the planning didn't allow it. Our product manager just said we aren't going to do this.

Dim3nsioneer, 2017-06-26:
But you would accept a proper pull request if someone from the community does it, right?

fieldOfView, 2017-06-26:
That is generally the case.

fieldOfView, 2017-06-26:
> But you would accept a proper pull request if someone from the community does it, right?

Let's test that theory.
It was fairly trivial to implement.

fieldOfView, 2017-06-28:
Feature has been implemented for Cura 2.7. This issue can be closed

closed 2017-06-28 by nallath

Dim3nsioneer, 2017-06-28:
Thanks to all who made this happening. Great.

Labern2, 2017-06-28:
This is awesome. Thanks.